Wiz-IO/wizio-pico #1: Trouble building examples

blackketter, 2021-02-23:
Trying to build the examples.  Source files compile linking fails 

This is on Intel MacBook Pro, PlatformIO 5.1.0, installed wizio-pico via command line:

`platformio platform install https://github.com/Wiz-IO/wizio-pico
`

Arduino examples fail linking with:

```
Processing pico (platform: wizio-pico; board: pico; framework: arduino; upload_port: E:/; monitor_port: COM34; monitor_speed: 115200)
------------------------------------------------------------------------------------------------------------------------------------------------------------
<<<<<<<<<<<< WIZIO-PICO 2021 Georgi Angelov >>>>>>>>>>>>
CONFIGURATION: https://docs.platformio.org/page/boards/wizio-pico/pico.html
PLATFORM: WizIO - Raspberry Pi Pico (1.0.0+sha.c9c8a28) (git+https://github.com/Wiz-IO/wizio-pico) > WizIO-PICO
HARDWARE: RP2040 133MHz, 256KB RAM, 2MB Flash
DEBUG: Current (uart) On-board (uart)
PACKAGES:
 - framework-wizio-pico 1.0.0+sha.b66c655 (git+https://github.com/Wiz-IO/framework-wizio-pico.git)
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
RASPBERRYPI PI PICO RP2040 ARDUINO
Heap Size: 65536
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 4 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
arm-none-eabi-g++ -o .pio/build/pico/ROM.elf -T /Users/dean/.platformio/packages/framework-wizio-pico/pico-sdk/src/rp2_common/pico_standard_link/memmap_default.ld -mcpu=cortex-m0plus -mthumb -Os -nostartfiles -mno-unaligned-access -Wall -Wfatal-errors -fno-use-cxa-atexit -fno-zero-initialized-in-bss -Xlinker --gc-sections -Wl,--gc-sections --entry=_entry_point .pio/build/pico/src/main.cpp.o -L.pio/build/pico .pio/build/pico/lib_arduino.a .pio/build/pico/lib_core.a .pio/build/pico/lib_variant.a .pio/build/pico/_arduino/libcommon.a .pio/build/pico/_arduino/librp2_common.a .pio/build/pico/lib_custom.a .pio/build/pico/_arduino/libboot2.a
.pio/build/pico/_arduino/librp2_common.a(runtime.c.o): In function `runtime_init':
runtime.c:(.text.runtime_init+0x80): undefined reference to `mutex_init'
collect2: error: ld returned 1 exit status
*** [.pio/build/pico/ROM.elf] Error 1
================================================================ [FAILED] Took 0.87 seconds ================================================================
```

Baremetal fails with:

```
<<<<<<<<<<<< WIZIO-PICO 2021 Georgi Angelov >>>>>>>>>>>>
CONFIGURATION: https://docs.platformio.org/page/boards/wizio-pico/pico.html
PLATFORM: WizIO - Raspberry Pi Pico (1.0.0+sha.c9c8a28) (git+https://github.com/Wiz-IO/wizio-pico) > WizIO-PICO
HARDWARE: RP2040 133MHz, 256KB RAM, 2MB Flash
DEBUG: Current (uart) On-board (uart)
PACKAGES:
 - framework-wizio-pico 1.0.0+sha.b66c655 (git+https://github.com/Wiz-IO/framework-wizio-pico.git)
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
RASPBERRYPI PI PICO RP2040 PICO-SDK
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
arm-none-eabi-gcc -o .pio/build/pico/ROM.elf -T /Users/dean/.platformio/packages/framework-wizio-pico/pico-sdk/src/rp2_common/pico_standard_link/memmap_default.ld -mcpu=cortex-m0plus -mthumb -Os -nostartfiles -mno-unaligned-access -Wall -Wfatal-errors -fno-use-cxa-atexit -fno-zero-initialized-in-bss -Xlinker --gc-sections -Wl,--gc-sections --entry=_entry_point .pio/build/pico/src/main.o -L.pio/build/pico .pio/build/pico/_sdk/libcommon.a .pio/build/pico/_sdk/librp2_common.a .pio/build/pico/lib_custom.a .pio/build/pico/_sdk/libboot2.a .pio/build/pico/_sdk/libstd.a
/Users/dean/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld: ERROR: Pico second stage bootloader must be 256 bytes in size
.pio/build/pico/_sdk/libstd.a(crt0.o): In function `hold_non_core0_in_bootrom':
(.reset+0x7c): undefined reference to `rom_func_lookup'
.pio/build/pico/_sdk/librp2_common.a(runtime.o): In function `runtime_init':
runtime.c:(.text.runtime_init+0x80): undefined reference to `mutex_init'
.pio/build/pico/_sdk/librp2_common.a(stdio.o): In function `stdout_serialize_begin':
stdio.c:(.text.stdout_serialize_begin+0xe): undefined reference to `mutex_try_enter'
stdio.c:(.text.stdout_serialize_begin+0x1e): undefined reference to `mutex_enter_blocking'
.pio/build/pico/_sdk/librp2_common.a(stdio.o): In function `stdout_serialize_end':
stdio.c:(.text.stdout_serialize_end+0x4): undefined reference to `mutex_exit'
.pio/build/pico/_sdk/libstd.a(syscall.o): In function `system_init':
syscall.c:(.text.system_init+0x2): undefined reference to `add_bootloader'
collect2: error: ld returned 1 exit status
*** [.pio/build/pico/ROM.elf] Error 1
================================================================ [FAILED] Took 1.09 seconds ================================================================
```

Wiz-IO, 2021-02-23:
looks like missing files...
...../pico-sdk/src/rp2_common/pico_bootrom/bootrom.c
...../pico-sdk/src/common/pico_sync/mutex.c

Clean and compile... look in log bootrom.o, mutex.o

blackketter, 2021-02-24:
Strange.  The bootrom.c and mutex.c files are there and appear to be built correctly:

```
-rw-r--r--  1 dean  staff  1255 Feb 20 08:02 .platformio/packages/framework-wizio-pico/pico-sdk/src/rp2_common/pico_bootrom/bootrom.c
-rw-r--r--  1 dean  staff  2126 Feb 20 08:02 .platformio/packages/framework-wizio-pico/pico-sdk/src/common/pico_sync/mutex.c
-rw-r--r--  1 dean  staff  3424 Feb 24 07:55 .pio/build/pico/_arduino/common/pico_sync/mutex.c.o
-rw-r--r--  1 dean  staff  1288 Feb 24 07:55 .pio/build/pico/_arduino/rp2_common/pico_bootrom/bootrom.c.o
```

Full verbose results attached:

[arduinoresults.txt](https://github.com/Wiz-IO/wizio-pico/files/6037271/arduinoresults.txt)
[baremetalresults.txt](https://github.com/Wiz-IO/wizio-pico/files/6037272/baremetalresults.txt)

Wiz-IO, 2021-02-24:
strange ... I test on two Wins 7 & 10 and all is ok
are you make only terminal builds?
Try with VSCode+PIO - visual IDE

Wiz-IO, 2021-03-04:
new version...

closed 2021-03-04 by Wiz-IO