Wiz-IO/wizio-pico #2: tinyusb required to build examples

blackketter, 2021-02-25:
Updated to latest sources, building examples fails with errors requiring `tusb.h`, like:

```
/Users/dean/.platformio/packages/framework-wizio-pico/pico-sdk/src/rp2_common/pico_stdio_usb/stdio_usb.c:8:10: fatal error: tusb.h: No such file or directory
```

Adding: `board_build.use_usb = 1` to platformio.ini files seems to fix it.  I believe that the intent is that USB is optional, so this shouldn't be necessary.

(Once added, the build still fails as reported in #1)

Wiz-IO, 2021-02-25:
sorry - The project is a work in progress and I do changes ( last change is before hour )
make this
https://github.com/Wiz-IO/wizio-pico#fast-uninstal
then install platform again
https://www.raspberrypi.org/forums/viewtopic.php?f=144&t=304882&e=1&view=unread#p1826770

USB example - create new baremetal project and compile default template
paste this files
https://github.com/raspberrypi/pico-examples/tree/master/usb/device/dev_hid_generic_inout
delete main.c 
renane dev_hid_generic_inout.c to main.c

in INI add 
board_build.use_usb = 1

and read this wiki
https://github.com/Wiz-IO/wizio-pico/wiki/BAREMETAL

Wiz-IO, 2021-02-25:
here is last changes
https://www.raspberrypi.org/forums/viewtopic.php?f=144&t=304882&e=1&view=unread#p1826770

Wiz-IO, 2021-03-04:
new version...

closed 2021-03-04 by Wiz-IO