Wiz-IO/wizio-pico #10: Examples not building

blackketter, 2021-03-04:
I'm trying to build the examples after the major updates (Thanks!), but with new errors.
This is with unmodified examples and a clean build using PlatformIO Core, version 5.1.0 on macOS via the command line.

The bare metal example is failing with:

```
dean@lucky13 pico-hello % pio run -v
Processing raspberry-pi-pico (platform: wizio-pico; board: raspberry-pi-pico; framework: baremetal; upload_port: E:/; monitor_port: COM34; monitor_speed: 115200; lib_deps: ; build_flags: -D PICO_STDIO_UART)
--------------------------------------------------------------------------------------------------------------------------------------------------------------
<<<<<<<<<<<< WIZIO - RASPBERRY PI PICO 2021 Georgi Angelov >>>>>>>>>>>>
CONFIGURATION: https://docs.platformio.org/page/boards/wizio-pico/raspberry-pi-pico.html
PLATFORM: WizIO - Raspberry Pi Pico (1.0.0+sha.15d6dbd) (git+https://github.com/Wiz-IO/wizio-pico) > WizIO - Raspberry Pi Pico
HARDWARE: RP2040 48MHz, 256KB RAM, 2MB Flash
DEBUG: Current (UF2-MSD) On-board (UF2-MSD, picoprobe-OpenOCD)
PACKAGES:
 - framework-wizio-pico 1.0.0+sha.bdd143f (git+https://github.com/Wiz-IO/framework-wizio-pico.git)
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
RASPBERRYPI PI PICO RP2040 PICO-SDK
FileNotFoundError: [Errno 2] No such file or directory: '/Users/dean/.platformio/packages/framework-wizio-pico/templates/config_autogen.h':
  File "/Users/dean/.platformio/penv/lib/python3.9/site-packages/platformio/builder/main.py", line 177:
    env.SConscript("$BUILD_SCRIPT")
  File "/Users/dean/.platformio/packages/tool-scons/scons-local-4.1.0/SCons/Script/SConscript.py", line 591:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/Users/dean/.platformio/packages/tool-scons/scons-local-4.1.0/SCons/Script/SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/Users/dean/.platformio/platforms/wizio-pico/builder/main.py", line 16:
    elf = env.BuildProgram()
  File "/Users/dean/.platformio/packages/tool-scons/scons-local-4.1.0/SCons/Util.py", line 658:
    return self.method(*nargs, **kwargs)
  File "/Users/dean/.platformio/penv/lib/python3.9/site-packages/platformio/builder/tools/platformio.py", line 61:
    env.ProcessProgramDeps()
  File "/Users/dean/.platformio/packages/tool-scons/scons-local-4.1.0/SCons/Util.py", line 658:
    return self.method(*nargs, **kwargs)
  File "/Users/dean/.platformio/penv/lib/python3.9/site-packages/platformio/builder/tools/platformio.py", line 127:
    env.BuildFrameworks(env.get("PIOFRAMEWORK"))
  File "/Users/dean/.platformio/packages/tool-scons/scons-local-4.1.0/SCons/Util.py", line 658:
    return self.method(*nargs, **kwargs)
  File "/Users/dean/.platformio/penv/lib/python3.9/site-packages/platformio/builder/tools/platformio.py", line 342:
    SConscript(env.GetFrameworkScript(f), exports="env")
  File "/Users/dean/.platformio/packages/tool-scons/scons-local-4.1.0/SCons/Script/SConscript.py", line 654:
    return method(*args, **kw)
  File "/Users/dean/.platformio/packages/tool-scons/scons-local-4.1.0/SCons/Script/SConscript.py", line 591:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/Users/dean/.platformio/packages/tool-scons/scons-local-4.1.0/SCons/Script/SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/Users/dean/.platformio/platforms/wizio-pico/builder/frameworks/sdk.py", line 13:
    m.dev_init(env, platform)
  File "/Users/dean/.platformio/platforms/wizio-pico/builder/frameworks/sdk-RP2040.py", line 16:
    dev_create_template(env)
  File "/Users/dean/.platformio/platforms/wizio-pico/builder/frameworks/common.py", line 44:
    do_copy(src, dst, "config_autogen.h" )
  File "/Users/dean/.platformio/platforms/wizio-pico/builder/frameworks/common.py", line 20:
    copyfile( join(src, name), join(dst, name) )
  File "/usr/local/Cellar/python@3.9/3.9.1_8/Frameworks/Python.framework/Versions/3.9/lib/python3.9/shutil.py", line 264:
    with open(src, 'rb') as fsrc, open(dst, 'wb') as fdst:
================================================================= [FAILED] Took 0.35 seconds =================================================================
dean@lucky13 pico-hello %
```

and the Arduino version fails with this: 
```
dean@lucky13 pico-hello % pio run
Processing raspberry-pi-pico (platform: wizio-pico; board: raspberry-pi-pico; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
<<<<<<<<<<<< WIZIO - RASPBERRY PI PICO 2021 Georgi Angelov >>>>>>>>>>>>
CONFIGURATION: https://docs.platformio.org/page/boards/wizio-pico/raspberry-pi-pico.html
PLATFORM: WizIO - Raspberry Pi Pico (1.0.0+sha.15d6dbd) > WizIO - Raspberry Pi Pico
HARDWARE: RP2040 48MHz, 256KB RAM, 2MB Flash
DEBUG: Current (UF2-MSD) On-board (UF2-MSD, picoprobe-OpenOCD)
[... 90 lines cut ...]
* Web  > https://platformio.org/lib/search?query=header:interface.h
*
*******************************************************************

 #include <interface.h>
          ^~~~~~~~~~~~~
compilation terminated.
*** [.pio/build/raspberry-pi-pico/arduino/arduino/Print.cpp.o] Error 1
/Users/dean/.platformio/packages/framework-wizio-pico/arduino/arduino/WString.cpp:23:10: fatal error: interface.h: No such file or directory

*******************************************************************
* Looking for interface.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:interface.h"
* Web  > https://platformio.org/lib/search?query=header:interface.h
*
*******************************************************************

 #include "interface.h"
          ^~~~~~~~~~~~~
compilation terminated.
*** [.pio/build/raspberry-pi-pico/arduino/arduino/WString.cpp.o] Error 1
================================================================= [FAILED] Took 0.55 seconds =================================================================
dean@lucky13 pico-hello %
```

Wiz-IO, 2021-03-04:
new version 1.0.1... reinstall platform
read readme and wiki

blackketter, 2021-03-04:
Thanks!  I had forgotten to update the framework-wizio-pico.  Sorry about that!

Arduino now builds fine, but there's another issue with the baremetal build and the unmodified platformio.ini:
```
Linking .pio/build/raspberry-pi-pico/APPLICATION.elf
.pio/build/raspberry-pi-pico/sdk/wizio/libpico.a(crt0.o): In function `hold_non_core0_in_bootrom':
(.reset+0x7c): undefined reference to `rom_func_lookup'
.pio/build/raspberry-pi-pico/sdk/libpico_stdio_uart.a(stdio_uart.o): In function `stdio_uart_out_chars':
stdio_uart.c:(.text.stdio_uart_out_chars+0x38): undefined reference to `uart_char_to_line_feed'
.pio/build/raspberry-pi-pico/sdk/libpico_stdio_uart.a(stdio_uart.o): In function `stdio_uart_init_full':
stdio_uart.c:(.text.stdio_uart_init_full+0x6): undefined reference to `uart_init'
collect2: error: ld returned 1 exit status
*** [.pio/build/raspberry-pi-pico/APPLICATION.elf] Error 1
```

Wiz-IO, 2021-03-04:
build_flags... enable stdio
https://github.com/Wiz-IO/wizio-pico/wiki/BAREMETAL#stdio--uart-usb-semihosting-

blackketter, 2021-03-04:
Thanks, I tried that too:
```
Linking .pio/build/raspberry-pi-pico/APPLICATION.elf
.pio/build/raspberry-pi-pico/sdk/wizio/libpico.a(crt0.o): In function `hold_non_core0_in_bootrom':
(.reset+0x7c): undefined reference to `rom_func_lookup'
.pio/build/raspberry-pi-pico/sdk/SDK/pico/libpico_stdio_usb.a(stdio_usb.o): In function `low_priority_worker_irq':
stdio_usb.c:(.text.low_priority_worker_irq+0x10): undefined reference to `tud_task'
.pio/build/raspberry-pi-pico/sdk/SDK/pico/libpico_stdio_usb.a(stdio_usb.o): In function `stdio_usb_in_chars':
stdio_usb.c:(.text.stdio_usb_in_chars+0x28): undefined reference to `tud_cdc_n_connected'
stdio_usb.c:(.text.stdio_usb_in_chars+0x40): undefined reference to `tud_cdc_n_available'
stdio_usb.c:(.text.stdio_usb_in_chars+0x4e): undefined reference to `tud_cdc_n_read'
.pio/build/raspberry-pi-pico/sdk/SDK/pico/libpico_stdio_usb.a(stdio_usb.o): In function `stdio_usb_out_chars':
stdio_usb.c:(.text.stdio_usb_out_chars+0x2a): undefined reference to `tud_cdc_n_connected'
stdio_usb.c:(.text.stdio_usb_out_chars+0x44): undefined reference to `tud_cdc_n_write_available'
stdio_usb.c:(.text.stdio_usb_out_chars+0x5a): undefined reference to `tud_cdc_n_write'
stdio_usb.c:(.text.stdio_usb_out_chars+0x60): undefined reference to `tud_task'
stdio_usb.c:(.text.stdio_usb_out_chars+0x66): undefined reference to `tud_cdc_n_write_flush'
stdio_usb.c:(.text.stdio_usb_out_chars+0x82): undefined reference to `tud_task'
stdio_usb.c:(.text.stdio_usb_out_chars+0x88): undefined reference to `tud_cdc_n_write_flush'
stdio_usb.c:(.text.stdio_usb_out_chars+0x8e): undefined reference to `tud_cdc_n_connected'
stdio_usb.c:(.text.stdio_usb_out_chars+0x98): undefined reference to `tud_cdc_n_write_available'
.pio/build/raspberry-pi-pico/sdk/SDK/pico/libpico_stdio_usb.a(stdio_usb.o): In function `stdio_usb_init':
stdio_usb.c:(.text.stdio_usb_init+0x8): undefined reference to `tusb_init'
collect2: error: ld returned 1 exit status
*** [.pio/build/raspberry-pi-pico/APPLICATION.elf] Error 1
========================================================= [FAILED] Took 0.95 seconds =========================================================
```
This is the content of the platformio.ini:

```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:raspberry-pi-pico]
platform = wizio-pico
board = raspberry-pi-pico
framework = baremetal

upload_port   = E:/
monitor_port  = COM34
monitor_speed = 115200

lib_deps =
        tinyusb

build_flags =
        -D PICO_STDIO_USB
        ;-D PICO_STDIO_UART
```

Wiz-IO, 2021-03-04:
what example? 
after hour will test

blackketter, 2021-03-04:
Thanks.  This is the `examples/baremetal/pico-hello` built with `pio run`.  Full build results:
```
dean@lucky13 pico-hello % ls
include/        platformio.ini  src/
dean@lucky13 pico-hello % cat platformio.ini
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:raspberry-pi-pico]
platform = wizio-pico
board = raspberry-pi-pico
framework = baremetal

upload_port   = E:/
monitor_port  = COM34
monitor_speed = 115200

lib_deps =
        tinyusb

build_flags =
        -D PICO_STDIO_USB
        ;-D PICO_STDIO_UART

dean@lucky13 pico-hello % pio run
Processing raspberry-pi-pico (platform: wizio-pico; board: raspberry-pi-pico; framework: baremetal)
--------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
<<<<<<<<<<<< WIZIO - RASPBERRY PI PICO 2021 Georgi Angelov >>>>>>>>>>>>
CONFIGURATION: https://docs.platformio.org/page/boards/wizio-pico/raspberry-pi-pico.html
PLATFORM: WizIO - Raspberry Pi Pico (1.0.1+sha.7bcf455) > WizIO - Raspberry Pi Pico
HARDWARE: RP2040 48MHz, 256KB RAM, 2MB Flash
DEBUG: Current (UF2-MSD) On-board (UF2-MSD, picoprobe-OpenOCD)
PACKAGES:
 - framework-wizio-pico 1.0.1+sha.b761da1
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
RASPBERRYPI PI PICO RP2040 PICO-SDK
  OPTIMIZATION: -Os
  HEAP: 2048
  BOOT: w25q080
  BINARY TYPE: default
  LINKER: memmap_default.ld
  ADDRESS: 0x10000000
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 5 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <tinyusb>
Building in release mode
Compiling .pio/build/raspberry-pi-pico/src/main.o
Compiling .pio/build/raspberry-pi-pico/lib87e/tinyusb/class/audio/audio_device.o
Compiling .pio/build/raspberry-pi-pico/lib87e/tinyusb/class/bth/bth_device.o
Compiling .pio/build/raspberry-pi-pico/lib87e/tinyusb/class/cdc/cdc_device.o
Compiling .pio/build/raspberry-pi-pico/lib87e/tinyusb/class/cdc/cdc_host.o
Compiling .pio/build/raspberry-pi-pico/lib87e/tinyusb/class/cdc/cdc_rndis_host.o
Compiling .pio/build/raspberry-pi-pico/lib87e/tinyusb/class/dfu/dfu_rt_device.o
Compiling .pio/build/raspberry-pi-pico/lib87e/tinyusb/class/hid/hid_device.o
Compiling .pio/build/raspberry-pi-pico/lib87e/tinyusb/class/hid/hid_host.o
Compiling .pio/build/raspberry-pi-pico/lib87e/tinyusb/class/midi/midi_device.o
Compiling .pio/build/raspberry-pi-pico/lib87e/tinyusb/class/msc/msc_device.o
Compiling .pio/build/raspberry-pi-pico/lib87e/tinyusb/class/msc/msc_host.o
Compiling .pio/build/raspberry-pi-pico/lib87e/tinyusb/class/net/net_device.o
[... 134 lines cut ...]
.pio/build/raspberry-pi-pico/sdk/wizio/libpico.a(crt0.o): In function `hold_non_core0_in_bootrom':
(.reset+0x7c): undefined reference to `rom_func_lookup'
.pio/build/raspberry-pi-pico/sdk/SDK/pico/libpico_stdio_usb.a(stdio_usb.o): In function `low_priority_worker_irq':
stdio_usb.c:(.text.low_priority_worker_irq+0x10): undefined reference to `tud_task'
.pio/build/raspberry-pi-pico/sdk/SDK/pico/libpico_stdio_usb.a(stdio_usb.o): In function `stdio_usb_in_chars':
stdio_usb.c:(.text.stdio_usb_in_chars+0x28): undefined reference to `tud_cdc_n_connected'
stdio_usb.c:(.text.stdio_usb_in_chars+0x40): undefined reference to `tud_cdc_n_available'
stdio_usb.c:(.text.stdio_usb_in_chars+0x4e): undefined reference to `tud_cdc_n_read'
.pio/build/raspberry-pi-pico/sdk/SDK/pico/libpico_stdio_usb.a(stdio_usb.o): In function `stdio_usb_out_chars':
stdio_usb.c:(.text.stdio_usb_out_chars+0x2a): undefined reference to `tud_cdc_n_connected'
stdio_usb.c:(.text.stdio_usb_out_chars+0x44): undefined reference to `tud_cdc_n_write_available'
stdio_usb.c:(.text.stdio_usb_out_chars+0x5a): undefined reference to `tud_cdc_n_write'
stdio_usb.c:(.text.stdio_usb_out_chars+0x60): undefined reference to `tud_task'
stdio_usb.c:(.text.stdio_usb_out_chars+0x66): undefined reference to `tud_cdc_n_write_flush'
stdio_usb.c:(.text.stdio_usb_out_chars+0x82): undefined reference to `tud_task'
stdio_usb.c:(.text.stdio_usb_out_chars+0x88): undefined reference to `tud_cdc_n_write_flush'
stdio_usb.c:(.text.stdio_usb_out_chars+0x8e): undefined reference to `tud_cdc_n_connected'
stdio_usb.c:(.text.stdio_usb_out_chars+0x98): undefined reference to `tud_cdc_n_write_available'
.pio/build/raspberry-pi-pico/sdk/SDK/pico/libpico_stdio_usb.a(stdio_usb.o): In function `stdio_usb_init':
stdio_usb.c:(.text.stdio_usb_init+0x8): undefined reference to `tusb_init'
collect2: error: ld returned 1 exit status
*** [.pio/build/raspberry-pi-pico/APPLICATION.elf] Error 1
================================================================= [FAILED] Took 1.20 seconds =================================================================
dean@lucky13 pico-hello %
```

Wiz-IO, 2021-03-04:
Windows 10

```
[env:raspberry-pi-pico]
platform = wizio-pico
board = raspberry-pi-pico
framework = baremetal
upload_port   = E:/
monitor_port  = COM34
monitor_speed = 115200
lib_deps = 
        tinyusb
build_flags = 
        -D PICO_STDIO_UART   
        -D PICO_STDIO_USB  
``` 
terminal:
**./pio run -d C:/Users/HP/Documents/PlatformIO/Projects/pico-hello**

```
Processing raspberry-pi-pico (platform: wizio-pico; board: raspberry-pi-pico; framework: baremetal)
---------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
<<<<<<<<<<<< WIZIO - RASPBERRY PI PICO 2021 Georgi Angelov >>>>>>>>>>>>
CONFIGURATION: https://docs.platformio.org/page/boards/wizio-pico/raspberry-pi-pico.html
PLATFORM: WizIO - Raspberry Pi Pico (1.0.1+sha.7bcf455) > WizIO - Raspberry Pi Pico
HARDWARE: RP2040 48MHz, 256KB RAM, 2MB Flash
DEBUG: Current (UF2-MSD) On-board (UF2-MSD, picoprobe-OpenOCD)
PACKAGES:
 - framework-wizio-pico 1.0.1+sha.b761da1
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
RASPBERRYPI PI PICO RP2040 PICO-SDK
  OPTIMIZATION: -Os
  HEAP: 2048
  BOOT: w25q080
  BINARY TYPE: default
  LINKER: memmap_default.ld
  ADDRESS: 0x10000000
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 6 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
DONE
Checking size .pio\build\raspberry-pi-pico\APPLICATION.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   4.7% (used 12192 bytes from 262144 bytes)
Flash: [          ]   1.8% (used 37876 bytes from 2097152 bytes)
=========================================================================================== [SUCCESS] Took 1.67 seconds 
```

Wiz-IO, 2021-03-04:
direct from examples folder

./pio run -d C:/Users/HP/.platformio/platforms/wizio-pico/examples/baremetal/pico-hello
...
=== [SUCCESS] Took 7.19 seconds ===

blackketter, 2021-03-04:
Very strange.  I'm building in the same directory:

 pio run -d ~/.platformio/platforms/wizio-pico/examples/baremetal/pico-hello

The platform and package versions seem to match.

Can you build with -v to help see if there are any differences in the build commands?

(Sorry if I'm not seeing something obvious here, I usually depend on platformio on handling this stuff perfectly! :)

```
dean@lucky13 ~ % pio run -vd ~/.platformio/platforms/wizio-pico/examples/baremetal/pico-hello
Processing raspberry-pi-pico (platform: wizio-pico; board: raspberry-pi-pico; framework: baremetal; upload_port: E:/; monitor_port: COM34; monitor_speed: 115200; lib_deps: ; build_flags: -D PICO_STDIO_UART)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
<<<<<<<<<<<< WIZIO - RASPBERRY PI PICO 2021 Georgi Angelov >>>>>>>>>>>>
CONFIGURATION: https://docs.platformio.org/page/boards/wizio-pico/raspberry-pi-pico.html
PLATFORM: WizIO - Raspberry Pi Pico (1.0.1+sha.7bcf455) (git+https://github.com/Wiz-IO/wizio-pico) > WizIO - Raspberry Pi Pico
HARDWARE: RP2040 48MHz, 256KB RAM, 2MB Flash
DEBUG: Current (UF2-MSD) On-board (UF2-MSD, picoprobe-OpenOCD)
PACKAGES:
 - framework-wizio-pico 1.0.1+sha.b761da1 (git+https://github.com/Wiz-IO/framework-wizio-pico.git)
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
RASPBERRYPI PI PICO RP2040 PICO-SDK
  OPTIMIZATION: -Os
  HEAP: 2048
  BOOT: w25q080
  BINARY TYPE: default
  LINKER: memmap_default.ld
  ADDRESS: 0x10000000
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 5 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
arm-none-eabi-gcc -o .pio/build/raspberry-pi-pico/src/main.o -c -mcpu=cortex-m0plus -mthumb -Os -fdata-sections -ffunction-sections -fno-strict-aliasing -fno-zero-initialized-in-bss -fsingle-precision-constant -Wall -Wfatal-errors -Wstrict-prototypes -Wno-unused-function -Wno-unused-but-set-variable -Wno-unused-variable -Wno-unused-value -Wno-discarded-qualifiers -mno-unaligned-access -mcpu=cortex-m0plus -mthumb -Os -fdata-sections -ffunction-sections -fno-strict-aliasing -fno-zero-initialized-in-bss -Wall -Wfatal-errors -Wno-unused-function -Wno-unused-but-set-variable -Wno-unused-variable -Wno-unused-value -mno-unaligned-access -DPLATFORMIO=50100 -DPICO_STDIO_UART -DPICO_ON_DEVICE=1 -DPICO_HEAP_SIZE=2048 -DCFG_TUSB_MCU=OPT_MCU_RP2040 -DSDK -Iinclude -Isrc -Isrc -Ilib -Iinclude -I/Users/dean/.platformio/packages/framework-wizio-pico/SDK/include -I/Users/dean/.platformio/packages/framework-wizio-pico/SDK/boards -I/Users/dean/.platformio/packages/framework-wizio-pico/SDK/pico/pico_stdio_uart/include src/main.c
arm-none-eabi-gcc -o .pio/build/raspberry-pi-pico/sdk/SDK/hardware/adc.o -c -mcpu=cortex-m0plus -mthumb -Os -fdata-sections -ffunction-sections -fno-strict-aliasing -fno-zero-initialized-in-bss -fsingle-precision-constant -Wall -Wfatal-errors -Wstrict-prototypes -Wno-unused-function -Wno-unused-but-set-variable -Wno-unused-variable -Wno-unused-value -Wno-discarded-qualifiers -mno-unaligned-access -mcpu=cortex-m0plus -mthumb -Os -fdata-sections -ffunction-sections -fno-strict-aliasing -fno-zero-initialized-in-bss -Wall -Wfatal-errors -Wno-unused-function -Wno-unused-but-set-variable -Wno-unused-variable -Wno-unused-value -mno-unaligned-access -DPLATFORMIO=50100 -DPICO_STDIO_UART -DPICO_ON_DEVICE=1 -DPICO_HEAP_SIZE=2048 -DCFG_TUSB_MCU=OPT_MCU_RP2040 -DSDK -Isrc -Ilib -Iinclude -I/Users/dean/.platformio/packages/framework-wizio-pico/SDK/include -I/Users/dean/.platformio/packages/framework-wizio-pico/SDK/boards -I/Users/dean/.platformio/packages/framework-wizio-pico/SDK/pico/pico_stdio_uart/include /Users/dean/.platformio/packages/framework-wizio-pico/SDK/hardware/adc.c
arm-none-eabi-gcc -o .pio/build/raspberry-pi-pico/sdk/SDK/hardware/claim.o -c -mcpu=cortex-m0plus -mthumb -Os -fdata-sections -ffunction-sections -fno-strict-aliasing -fno-zero-initialized-in-bss -fsingle-precision-constant -Wall -Wfatal-errors -Wstrict-prototypes -Wno-unused-function -Wno-unused-but-set-variable -Wno-unused-variable -Wno-unused-value -Wno-discarded-qualifiers -mno-unaligned-access -mcpu=cortex-m0plus -mthumb -Os -fdata-sections -ffunction-sections -fno-strict-aliasing -fno-zero-initialized-in-bss -Wall -Wfatal-errors -Wno-unused-function -Wno-unused-but-set-variable -Wno-unused-variable -Wno-unused-value -mno-unaligned-access -DPLATFORMIO=50100 -DPICO_STDIO_UART -DPICO_ON_DEVICE=1 -DPICO_HEAP_SIZE=2048 -DCFG_TUSB_MCU=OPT_MCU_RP2040 -DSDK -Isrc -Ilib -Iinclude -I/Users/dean/.platformio/packages/framework-wizio-pico/SDK/include -I/Users/dean/.platformio/packages/framework-wizio-pico/SDK/boards -I/Users/dean/.platformio/packages/framework-wizio-pico/SDK/pico/pico_stdio_uart/include /Users/dean/.platformio/packages/framework-wizio-pico/SDK/hardware/claim.c
arm-none-eabi-gcc -o .pio/build/raspberry-pi-pico/sdk/SDK/hardware/clocks.o -c -mcpu=cortex-m0plus -mthumb -Os -fdata-sections -ffunction-sections -fno-strict-aliasing -fno-zero-initialized-in-bss -fsingle-precision-constant -Wall -Wfatal-errors -Wstrict-prototypes -Wno-unused-function -Wno-unused-but-set-variable -Wno-unused-variable -Wno-unused-value -Wno-discarded-qualifiers [... cut ...]
stdio_uart.c:(.text.stdio_uart_init_full+0x6): undefined reference to `uart_init'
collect2: error: ld returned 1 exit status
*** [.pio/build/raspberry-pi-pico/APPLICATION.elf] Error 1
================================================================================================================================================ [FAILED] Took 1.58 seconds ================================================================================================================================================
dean@lucky13 ~ %
```

Wiz-IO, 2021-03-04:
my PIO is not in PATH and I pio run from 
C:\Users\HP\.platformio\penv\Scripts 
with -d argument path to project

log is inside
[pico-hello.zip](https://github.com/Wiz-IO/wizio-pico/files/6086731/pico-hello.zip)

Wiz-IO, 2021-03-04:
it is PIO "bug" ... -Wl,--start-group / -Wl,--end-group

https://github.com/platformio/platformio-core/issues/3674

blackketter, 2021-03-04:
Thank you, @Wiz-IO!

I also tried building under Linux (on a raspi 4, running raspian), and was able to build both pico-hello examples without error.  So that's good!

Not sure what's different on macOS.
I'm using the equivalent pio executable on macOS:  ~/.platformio/penv/bin/pio. Same result with an explicit path.

Comparing the verbose build commands I don't see any differences at all before the final build of the APPLICATION.elf.  The bootrom.o and libSDK.a files are the same size.

The final command that succeeds on linux is:

arm-none-eabi-gcc -o .pio/build/raspberry-pi-pico/APPLICATION.elf -T /home/dean/.platformio/packages/framework-wizio-pico/SDK/pico/pico_standard_link/memmap_default.ld -mcpu=cortex-m0plus -mthumb -Os -nostartfiles -mno-unaligned-access -Wall -Wfatal-errors -fno-use-cxa-atexit -fno-zero-initialized-in-bss -Xlinker --gc-sections -Wl,--gc-sections --entry=_entry_point .pio/build/raspberry-pi-pico/src/main.o -L.pio/build/raspberry-pi-pico -L/home/dean/.platformio/packages/framework-wizio-pico/library **-Wl,--start-group** -lm -lgcc .pio/build/raspberry-pi-pico/sdk/libSDK.a .pio/build/raspberry-pi-pico/sdk/wizio/libpico.a .pio/build/raspberry-pi-pico/sdk/wizio/libboot2.a .pio/build/raspberry-pi-pico/sdk/libpico_stdio_uart.a **-Wl,--end-group**

and fails on macOS is:

arm-none-eabi-gcc -o .pio/build/raspberry-pi-pico/APPLICATION.elf -T /home/dean/.platformio/packages/framework-wizio-pico/SDK/pico/pico_standard_link/memmap_default.ld -mcpu=cortex-m0plus -mthumb -Os -nostartfiles -mno-unaligned-access -Wall -Wfatal-errors -fno-use-cxa-atexit -fno-zero-initialized-in-bss -Xlinker --gc-sections -Wl,--gc-sections --entry=_entry_point .pio/build/raspberry-pi-pico/src/main.o -L.pio/build/raspberry-pi-pico -L/home/dean/.platformio/packages/framework-wizio-pico/library -lm -lgcc .pio/build/raspberry-pi-pico/sdk/libSDK.a .pio/build/raspberry-pi-pico/sdk/wizio/libpico.a .pio/build/raspberry-pi-pico/sdk/wizio/libboot2.a .pio/build/raspberry-pi-pico/sdk/libpico_stdio_uart.a
 
If you look closely there are options "-Wl,--start-group" and "-Wl,--end-group" in linux and windows, but not Mac.

I don't know where this difference comes from. Do you have any idea?

Wiz-IO, 2021-03-04:
some problem with OS X & GCC
read pio issue .... Ivan say: 
try `pio upgrade --dev`

blackketter, 2021-03-04:
upgraded via `pio upgrade --dev` to 5.1.1a3.

The `-Wl,--start-group` and `-Wl,--start-group` are still missing. 👎

closed 2021-03-31 by Wiz-IO