Wiz-IO/wizio-pico #11: Automatic preprocessing of Arduino .ino to .cpp files fail

blackketter, 2021-03-04:
Building a simple blink.ino sketch gives this error:

```
Processing raspberry-pi-pico (platform: wizio-pico; board: raspberry-pi-pico; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
<<<<<<<<<<<< WIZIO - RASPBERRY PI PICO 2021 Georgi Angelov >>>>>>>>>>>>
CONFIGURATION: https://docs.platformio.org/page/boards/wizio-pico/raspberry-pi-pico.html
PLATFORM: WizIO - Raspberry Pi Pico (1.0.1+sha.7bcf455) > WizIO - Raspberry Pi Pico
HARDWARE: RP2040 48MHz, 256KB RAM, 2MB Flash
DEBUG: Current (UF2-MSD) On-board (UF2-MSD, picoprobe-OpenOCD)
PACKAGES:
 - framework-wizio-pico 1.0.1+sha.b761da1
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
Converting Blink.ino
clang: error: unknown argument: '-fpreprocessed'
*** Error 1
```

It seems that the default CC (Clang on macOS) is being used.

CXX needs to be specified for the pico, probably arm-none-eabi-g++

File compiles fine if renamed to .cpp (and adding necessary Arduino.h include).

Wiz-IO, 2021-03-05:
PIO not support .ino
`File compiles fine if renamed to .cpp (and adding necessary Arduino.h include).`

blackketter, 2021-03-05:
> PIO not support .ino
I don't think this is true.  I have several platformio projects that use imported Arduino .ino files which are automatically processed to .cpp as an intermediate stage.  This works with every other platform and board I've tested. 

It's not a big deal for me, but I think that the Pico behavior should be consistent with the other platforms, this will help new users migrating from the Arduino tools.  I'll try to create a pull request.

Wiz-IO, 2021-03-05:
can you compile pioasm for OS X
https://github.com/raspberrypi/pico-sdk/tree/master/tools/pioasm

in folder pioasm
```
cmake .
make

```
IF/THEN just change the end of main.cpp with:

```
    else
    {
        std::cerr << "\n[PIO-ASM] SRC: " << input << "\n";
        std::cerr << "[PIO-ASM] DST: " << output << "\n";
        res = pioasm.generate(oformat, input, output, options);
    }
    if (res == 0)
        std::cerr << "[PIO-ASM] Done\n";
    return res;
}
```

blackketter, 2021-03-05:
Made this change:

```
diff --git a/tools/pioasm/main.cpp b/tools/pioasm/main.cpp
index 1c86bea..c16551a 100644
--- a/tools/pioasm/main.cpp
+++ b/tools/pioasm/main.cpp
@@ -94,7 +94,11 @@ int main(int argc, char *argv[]) {
         std::cerr << std::endl;
         usage();
     } else {
+        std::cerr << "\n[PIO-ASM] SRC: " << input << "\n";
+        std::cerr << "[PIO-ASM] DST: " << output << "\n";
         res = pioasm.generate(oformat, input, output, options);
     }
+    if (res == 0)
+        std::cerr << "[PIO-ASM] Done\n";
     return res;
 }
\ No newline at end of file
```
It generated this:
[pioasm.zip](https://github.com/Wiz-IO/wizio-pico/files/6091805/pioasm.zip)

Wiz-IO, 2021-03-05:
thanks
run this in python terminal ... need result strings
```
from platform import system, machine
print("SYSTEM", system())
print("MACHINE", machine())
```

blackketter, 2021-03-05:
>>> from platform import system, machine
>>> print("SYSTEM", system())
SYSTEM Darwin
>>> print("MACHINE", machine())
MACHINE x86_64

Wiz-IO, 2021-03-05:
THANKS !
if you reinstall platform - must support PIO-ASM

example
[a_pio_asm.zip](https://github.com/Wiz-IO/wizio-pico/files/6091904/a_pio_asm.zip)

Wiz-IO, 2021-03-05:
.... and delete **ws2812.pio.h** from project :)

if is compiled ... pioasm is skipped

blackketter, 2021-03-05:
No problem!

I don't quite understand what you mean by "if you reinstall platform - must support PIO-ASM".  Can you elaborate?

Wiz-IO, 2021-03-05:
:) sorry for my English
I updateд platform to support and pioasm

https://github.com/Wiz-IO/wizio-pico/blob/main/builder/main.py#L17
https://github.com/Wiz-IO/tool-wizio-pico

blackketter, 2021-03-05:
No problem.  Thanks for the excellent work!

closed 2021-03-24 by Wiz-IO