Wiz-IO/wizio-pico #24: baremetal linking error on Mac

deladriere, 2021-03-08:
when building I get this error 

```
Processing raspberry-pi-pico (platform: wizio-pico; board: raspberry-pi-pico; framework: baremetal)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
<<<<<<<<<<<< WIZIO - RASPBERRY PI PICO 2021 Georgi Angelov >>>>>>>>>>>>
CONFIGURATION: https://docs.platformio.org/page/boards/wizio-pico/raspberry-pi-pico.html
PLATFORM: WizIO - Raspberry Pi Pico (1.0.1+sha.e00cf3a) > WizIO - Raspberry Pi Pico
HARDWARE: RP2040 48MHz, 256KB RAM, 2MB Flash
DEBUG: Current (UF2-MSD) On-board (UF2-MSD, picoprobe-OpenOCD)
PACKAGES: 
 - framework-wizio-pico 1.0.1+sha.337aea4 
 - tool-wizio-pico 1.0.0+sha.83425b1 
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)

SDK RASPBERRYPI PI PICO RP2040 ( PICO - SDK110 )
  - OPTIMIZATION : -Os
  - HEAP         : 2048
  - STACK        : 2048
  - BOOT         : w25q080
  - BINARY TYPE  : default [ memmap_default.ld   0x10000000 ]

LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 5 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Linking .pio/build/raspberry-pi-pico/APPLICATION.elf
.pio/build/raspberry-pi-pico/sdk/wizio/libpico.a(crt0.o): In function `hold_non_core0_in_bootrom':
(.reset+0x7c): undefined reference to `rom_func_lookup'
collect2: error: ld returned 1 exit status
*** [.pio/build/raspberry-pi-pico/APPLICATION.elf] Error 1
```

```
[env:raspberry-pi-pico]
platform = wizio-pico
board = raspberry-pi-pico
framework = baremetal

upload_port  = PicoDrive:\
monitor_port  = COM1
monitor_speed = 115200
```

Wiz-IO, 2021-03-08:
https://github.com/Wiz-IO/wizio-pico/issues/10#issuecomment-790981999

deladriere, 2021-03-14:
Using  pio upgrade --dev
but still having the same error

Wiz-IO, 2021-03-14:
waiting for an answer from PlatformIO ...

blackketter, 2021-03-19:
Any news on this?

Wiz-IO, 2021-03-19:
still no :(

tsandmann, 2021-03-23:
As a workaround until this is fixed in PlatformIO you can add `extra_scripts = extra_script.py` to *platformio.ini* and create a simple script file *extra_script.py* with the following contents:

```python
Import("env")

flags = env["_LIBFLAGS"]
if not "-Wl,--start-group" in flags:
    env.Replace(_LIBFLAGS = ["-Wl,--start-group", flags, "-Wl,--end-group"])
```
to correct the linker command.

Wiz-IO, 2021-03-23:
if this work will integrate in builder
please test ... enter code here
https://github.com/Wiz-IO/wizio-pico/blob/main/builder/frameworks/common.py#L231
after **add_boot(env)**

Wiz-IO, 2021-03-23:
start/stop not exist in builder ... they is generated on next stage, after > Building in release mode

`+++++++++++++++++++++
${_stripixes(LIBLINKPREFIX, LIBS, LIBLINKSUFFIX, LIBPREFIXES, LIBSUFFIXES, __env__)}
+++++++++++++++++++++
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 16 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
---------------------
-Wl,--start-group ${_stripixes(LIBLINKPREFIX, LIBS, LIBLINKSUFFIX, LIBPREFIXES, LIBSUFFIXES, __env__)} -Wl,--end-group
---------------------`

tsandmann, 2021-03-23:
Yes, `extra_scripts` are executed after *"Building in release mode"*.

The root cause of the problem is that `env.PioPlatform().is_embedded()` returns `true` only if the packages list ([platform.json](../blob/master/platform.json)) contains a package of *type* **uploader** 🙈. That doesn't really make sense to me, but anyway, it's easy to "fix":
add a dummy uploader tool like
```json
"uf2": {
  "type": "uploader",
  "optional": true
}
```
to [platform.json:46](../blob/master/platform.json#L46) and the linker command is setup correctly on macOS.

Most embedded platforms specify an upload or debugger tool in their packages list and therefore don't have this issue...

Wiz-IO, 2021-03-23:
https://github.com/Wiz-IO/wizio-pico/blob/main/platform.json#L46

and now OS X is OK?

tsandmann, 2021-03-23:
Yes, it's fine now 👍

Wiz-IO, 2021-03-23:
Thanks !!!

beep @blackketter

tsandmann, 2021-03-24:
FYI: For the "official" solution see https://github.com/platformio/platformio-core/issues/3674#issuecomment-806032957.

deladriere, 2021-03-25:
Still not working for me 

```
Processing raspberry-pi-pico (platform: wizio-pico; board: raspberry-pi-pico; framework: baremetal)
-----------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
<<<<<<<<<<<< WIZIO - RASPBERRY PI PICO 2021 Georgi Angelov >>>>>>>>>>>>
CONFIGURATION: https://docs.platformio.org/page/boards/wizio-pico/raspberry-pi-pico.html
PLATFORM: WizIO - Raspberry Pi Pico (1.0.4+sha.28ab2b0) > WizIO - Raspberry Pi Pico
HARDWARE: RP2040 125MHz, 256KB RAM, 2MB Flash
DEBUG: Current (UF2-MSD) On-board (UF2-MSD, picoprobe-OpenOCD)
PACKAGES: 
 - framework-wizio-pico 1.0.4+sha.69aeff9 
 - tool-wizio-pico 1.0.0+sha.76bba42 
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)

```

```
Indexing .pio/build/raspberry-pi-pico/sdk/libSDK.a
Linking .pio/build/raspberry-pi-pico/APPLICATION.elf
.pio/build/raspberry-pi-pico/src/main.o: In function `main':
main.cpp:(.text.startup.main+0x6): undefined reference to `stdio_init_all'
.pio/build/raspberry-pi-pico/sdk/libwizio.a(syscall.o): In function `_write_r':
syscall.c:(.text._write_r+0x12): undefined reference to `_write'
.pio/build/raspberry-pi-pico/sdk/libwizio.a(syscall.o): In function `_read_r':
syscall.c:(.text._read_r+0x12): undefined reference to `_read'
collect2: error: ld returned 1 exit status
*** [.pio/build/raspberry-pi-pico/APPLICATION.elf] Error 1
================================================ [FAILED] Took 4.79 seconds ================================================
The terminal process "platformio 'run'" terminated with exit code: 1.
```

Wiz-IO, 2021-03-25:
that's exactly what I'm fixing now

workaround:
`build_flags = -D PICO_STDIO_UART`

deladriere, 2021-03-25:
@Wiz-IO Ok it works! 
can we upload directly? Now I need to drag the uf2 manually to the Pico

Wiz-IO, 2021-03-25:
what is the Mass Storage Device path name of picoboard on macOS

upload_port  = PicoDrive:\
maybe is /Volumes/<NameOfPicoDrive>

Wiz-IO, 2021-03-25:
[image]

deladriere, 2021-03-25:
thanks !! for me  it works with :
upload_port = /Volumes/RPI-RP2/:
(if I manually set the pico to bootloader mode)
Can we avoid the unplug - replug cycle ?

Wiz-IO, 2021-03-25:
what mean:
if I manually set the pico to bootloader mode
Can we avoid the unplug - replug cycle ?

deladriere, 2021-03-25:
It works only if I 
1 . plug the pico while holding the bootsel button so it mounts as a disk
2. I can then upload, the code runs  and the pico it ejected 
3. if I upload again it says : *** [upload] /Volumes/RPI-RP2/:APPLICATION.uf2: No such file or directory

If I want to upload I need to unplug the pico and redo step 1

Wiz-IO, 2021-03-25:
here is the "problem" :)

from code: reset_usb_boot(0,0);
and look in SDK manual 

look and this
https://www.raspberrypi.org/blog/how-to-add-a-reset-button-to-your-raspberry-pi-pico/

deladriere, 2021-03-25:
Ok thanks: I have installed a switch between run and gnd 
I wish this could be handled via the code (I like to update the code inside a device from the USB without having to open it)

RocketManRC, 2021-03-25:
This is a great project and really coming along nicely!

I had success building with this define too:

build_flags = -D PICO_STDIO_UART

which gives output to the USB serial port.

It also allows a project with the Arduino framework to build and upload successfully but there is no serial output to USB (I did not try the UART).

Wiz-IO, 2021-03-25:
re-install platform - test please v 1.0.5

build_flags = -D **PICO_STDIO_USB** must redirect printf() to Virtual USB COM Port ( no idea abuot macOS - on Win10 work )

deladriere, 2021-03-25:
I am on macOS
Printf works fine
here is my plaftformio.ini

```
[env:raspberry-pi-pico]
platform = wizio-pico
board = raspberry-pi-pico
framework = baremetal

build_flags = -D PICO_STDIO_UART
            -D PICO_STDIO_USB

upload_port = /Volumes/RPI-RP2/:
monitor_port = /dev/cu.usbmodem0000000000001
monitor_speed = 115200
```

blackketter, 2021-03-25:
I am unable to get the Printf to work on macOS with the flags above in Arduino.  The serial port appears and I can monitor it, but there's no output.  Should the flags be different for Arduino vs baremetal?

Wiz-IO, 2021-03-25:
for Arduino
Serial.begin(115200, **true**); // true - retarget std printf() 

https://github.com/Wiz-IO/wizio-pico/wiki/ARDUINO#stdio--serial-

baremetal
https://github.com/Wiz-IO/wizio-pico/wiki/BAREMETAL#ini-example

Wiz-IO, 2021-03-25:
on Arduino this will work if Serial is not used
```
build_flags = -D PICO_STDIO_UART
```
[image]

Wiz-IO, 2021-03-25:
yep :)
remove this 4 lines [ 138, 139, 140, 141 ]
https://github.com/Wiz-IO/framework-wizio-pico/blob/main/arduino/cores/RP2040/SerialUart.h#L138

Wiz-IO, 2021-03-25:
[image]

Wiz-IO, 2021-03-26:
**offtopic**
how is Linux path for: ( as example/doc )
- upload_port  = ?
- monitor_port  = ?

https://github.com/Wiz-IO/wizio-pico/wiki/COMMON#platformioini--minimal-

blackketter, 2021-03-26:
On raspian the upload port is:

/media/{username}/RPI-RP2

However, I'm not seeing a new serial device on raspian!  (That same board does appear as /dev/cu.usbmodem0000000000001 on macOS.)

blackketter, 2021-03-26:
I do see a new device at /dev/bus/usb/001/010 but that doesn't appear as a serial port.  Apparently, it _should_ appear as something like /dev/ttyACM<n>

Wiz-IO, 2021-03-26:
no idea ... documentation say: must be /dev/ttyACM<n>

btw: I "conect" pico  - lwip with interface over usb :) ... ping 17mS

tsandmann, 2021-04-01:
> Ok thanks: I have installed a switch between run and gnd
> I wish this could be handled via the code (I like to update the code inside a device from the USB without having to open it)

Maybe this can be integrated into the PIO upload command to allow updates without an extra reset-button or the annoying cable-unplugging: https://www.raspberrypi.org/forums/viewtopic.php?f=145&t=305458&p=1833379#p1833379

Wiz-IO, 2021-04-01:
I will look into it ... any ideas are welcome

Wiz-IO, 2021-04-01:
work fine, will update soon...
just need to find how to  **get("monitor_port")**

Wiz-IO, 2021-04-02:
re-install and test

not need **upload_port** in INI ( will try flash all Pico-s USB connected )

IF **PICO_STDIO_USB** is used, the uploader will try to reset to boot

vladkozlov69, 2021-04-02:
@Wiz-IO looks like it loses trailing slash now on Mac:

> upload_port = /Volumes/RPI-RP2/

> Uploading...
  Converting to UF2
  Start address: 0x10000000
  FamilyID: 0xe48bff56
  Wrote 89600 bytes to /Users/vkozlov/Documents/PlatformIO/Projects/test_2040/.pio/build/raspberry-pi-pico/ARDUINO.uf2
Flashing /Volumes/RPI-RP2 (RPI-RP2)
*** [upload] /Volumes/RPI-RP2ARDUINO.uf2: Operation not permitted

Wiz-IO, 2021-04-02:
huh... wiil fix after hour

Wiz-IO, 2021-04-02:
try fix your file
**user**\.platformio\platforms\wizio-pico\builder\frameworks\**uf2conv.py**
https://github.com/Wiz-IO/wizio-pico/blob/main/builder/frameworks/uf2conv.py#L341

vladkozlov69, 2021-04-02:
Yes, now it works as expected, thanks!

Wiz-IO, 2021-04-02:
BTW on WIN work without slash
pls try with join()

tsandmann, 2021-04-02:
Works great, thanks!

FYI: I had to increase [time.sleep(1.0)](https://github.com/Wiz-IO/wizio-pico/blob/main/builder/frameworks/uf2conv.py#L329) to 
```python
time.sleep(2.0) # Windows - AutoPlay
```
to make it work reliable on macOS.

Wiz-IO, 2021-04-02:
https://github.com/Wiz-IO/wizio-pico/blob/main/builder/frameworks/uf2conv.py#L329

deladriere, 2021-04-09:
@Wiz-IO The upload works great on macOS ! no need to play with the BOOTSEL button
Thanks a lot

closed 2021-05-10 by deladriere