Wiz-IO/wizio-pico #33: USB Flag Causes Build Errors

Kuroiyatsu, 2021-03-14:
First, thanks for this awesome work you've done. 

So when I have my platformio.ini file set as below, I receive the following errors when building: 

[image]

[image]

Kuroiyatsu, 2021-03-14:
I should also note that when I remove the build flag, the error goes away.

Wiz-IO, 2021-03-14:
Hi, please re-install platform and test again 
I make changes every day
if error - send me src example for debug

Wiz-IO, 2021-03-14:
maybe you found bug in the new SDK 1.1.0
these includes is missing
[image]

Kuroiyatsu, 2021-03-14:
Oh interesting. Yeah I reinstalled and received the same issue. That would make sense now.

Wiz-IO, 2021-03-14:
and for me give this errors and PC can not install driver

if you have original pico-sdk - please test

Kuroiyatsu, 2021-03-14:
It looks like I have 1.1.0

Wiz-IO, 2021-03-14:
usb stdio & sdk 1.0.1 - work for me
switch SDK in ini
`
board_build.sdk = SDK101 ; default is SDK110`

Kuroiyatsu, 2021-03-15:
That worked for me. 

Thanks again for the hard work!

closed 2021-03-15 by Kuroiyatsu