Wiz-IO/wizio-pico #43: F_CPU should be defined for CPP flags

blackketter, 2021-03-20:
The C++ build flags should include -DF_CPU for the default CPU speed, which I believe should be -DF_CPU=48000000

Wiz-IO, 2021-03-20:
ok will add 
this days will update platform to 1.0.3

Wiz-IO, 2021-03-20:
isn't it 125MHz by default

[The default system clock frequency for SDK is 125 MHz.](https://datasheets.raspberrypi.org/pico/raspberry-pi-pico-c-sdk.pdf) page 44

`uint f_clk_sys = frequency_count_khz(CLOCKS_FC0_SRC_VALUE_CLK_SYS);`

blackketter, 2021-03-20:
You may be right, I was only looking at the chip specification.

blackketter, 2021-03-20:
In any case, if the board_build.f_cpu is set in the platformio.ini file this should be reflected in the compilation flag...

https://docs.platformio.org/en/latest/projectconf/section_env_platform.html?highlight=board_build.f_cpu#board-build-f-cpu

Wiz-IO, 2021-03-22:
version 1.0.3

F_CPU is set on system_init() or after setCpuFrequency()
https://github.com/Wiz-IO/framework-wizio-pico/blob/main/arduino/cores/RP2040/interface.h#L64

closed 2021-03-23 by Wiz-IO