Wiz-IO/wizio-pico #63: PIOASM not working

hugoferreira, 2021-03-31:
From what I've seen, `pioasm` support was added in a0e41397f445566efca7b26f16af4aa2416aa99b with an example in a0e41397f445566efca7b26f16af4aa2416aa99b. It was also upgrade to support multiple files in bfe4a4a1494a93825f20beb76908901dd6266511. It seems all one needs to do is to add something like:

```
board_build.pio_src = src/blink.pio
```

... to `platformio.ini`, and a corresponding `include/blink.pio.h` would be created. However, #42 deleted all PIO examples, and I can't get them to work again. Was there a problem with `pioasm`?

Wiz-IO, 2021-04-01:
Hi re-install platform to 1.0.5
 new pio "key" is
```ini
board_build.pio = 
    src/ws2812.pio
    ;other pio file
```
https://github.com/Wiz-IO/wizio-pico/blob/main/examples/arduino/pico-pioasm-ws2812/platformio.ini#L16

I will rewrite the old examples for 1.0.5

hugoferreira, 2021-04-01:
It's working, although on macOS I had to `chmod +x` the `pioasm` installed in `.platformio/packages/tool-wizio-pico/darwin_x86_64`. Thx!

Wiz-IO, 2021-04-01:
no idea about mac :) not have... can not test

closed 2021-04-01 by Wiz-IO