Wiz-IO/wizio-pico #64: Could not load recent projects

Bellerophone, 2021-04-02:
After install (and reinstall) projects no longer load in PIO Home.
Made a bug report at PIO repo (https://github.com/platformio/platformio-home/issues/2395)

Server error: {"type":"AttributeError","args":["'WiziopicoPlatform' object has no attribute 'monitor_port'"],"message":"'WiziopicoPlatform' object has no attribute 'monitor_port'"}

Windows 10 x64
Core 5.1.1
Home 3.3.4

Wiz-IO, 2021-04-02:
now is version 1.0.5

Wiz-IO, 2021-04-02:
this **monitor_port** is used for reset to boot via USB stdio without button click

Wiz-IO, 2021-04-02:
I test fresh platform copy on WIN-10, can not see this error

closed 2021-04-02 by Wiz-IO

Bellerophone, 2021-04-02:
It magically fixed itself after yet another reinstall. Also, PIO Home reported 1.0.4 but the packages tab reports 1.0.5 weird.
Anyways, thank you

Wiz-IO, 2021-04-02:
sorry... The project is a work in progress and I make fixes/changes ... etc 
PIO load/install platform from this git ... I not have "money" to support file storage for the platform