Wiz-IO/wizio-pico #70: How to printf int64 with baremetal?

zapta, 2021-05-13:
According to users of the SDK, it comes out of the box with printf support for int64.  How can this be done with the baremetal environment?  I tried -DPICO_PRINTF_SUPPORT_LONG_LONG  and also tried PICO_PRINTF_PICO but both didn't work. I think that ideally the 'baremetal' version should come out of the box with the same defaults as the original C++ SDK.

My platformio.ini  https://pastebin.com/FThQg2J2

Wiz-IO, 2021-05-13:
"baremetal" is just a name of platform and use last release ( 1.1.2 ) of pico SDK

I may have missed something ... no idea
there must be descriptions of all the definitions somewhere, please ask in pico git or forum

Wiz-IO, 2021-05-13:
platformio.ini
**board_build.nano = disable**

by default is enabled: -specs=nano.specs

closed 2021-05-14 by Wiz-IO