Wiz-IO/wizio-pico #74: EEPROM Read Write

fluffymadness, 2021-05-28:
I'm not sure when/if it will become part of the official arduino ide
since this seems to be a fork, but this could
be quite useful to integrate
https://github.com/earlephilhower/arduino-pico/tree/a1c9dfe84d3c5070a2a7828618ebf856f84e6b9f/libraries/EEPROM

Wiz-IO, 2021-05-28:
are you want to integrate this library to wizio-pico?
copy / paste EEPROM in new_project/lib and test
if work will add in next update...

fluffymadness, 2021-05-28:
It builds when adding the following additional files from the repository

RP2040.h
CoreMutex.h

However I couldn't test it so far, because I still have to solve the issue with the correct flash address.
There is 
```
extern "C" uint8_t _EEPROM_start;
EEPROMClass::EEPROMClass(void)    : _sector(&EEPROM_start) {}
```
in the eeprom.cpp

searching for the _EEPROM_start value in the repository gave me the following:

```
lib/memmap_default.ld
    SCRATCH_X(rwx) : ORIGIN = 0x20040000, LENGTH = 4k
    SCRATCH_Y(rwx) : ORIGIN = 0x20041000, LENGTH = 4k
}
PROVIDE ( _EEPROM_start = __EEPROM_START__ );
```

and

```
/boards.txt
rpipico.menu.flash.2097152_0.build.eeprom_start=270528512
rpipico.menu.flash.2097152_0.build.fs_start=270528512
rpipico.menu.flash.2097152_65536.build.flash_length=2027520
rpipico.menu.flash.2097152_65536.build.eeprom_start=270528512

```
The documenation I linked above says

Wiz-IO, 2021-05-28:
https://github.com/Wiz-IO/framework-wizio-pico/blob/main/SDK/pico/pico_standard_link/memmap_default.ld#L234

something like that
```
    .flash_end : {
        . = ALIGN(4096);
        ._EEPROM_start = .;
        . = . + 4096;
        __flash_binary_end = .;
    } > FLASH
```

fluffymadness, 2021-05-29:
I made the edits in the file, that you suggested, but still haven't figured out how to 
get a value for extern "C" uint8_t _EEPROM_start in EEPROM.cpp ?
Can you please explain a bit in more detail how
I can get the correct address from the changed memmap_default.ld
so that it is applied to the variable in EEPROM.cpp ?

Meanwhile I took another look and the lib and I think it could be also built baremetal.
I replaced the interrupt disabled and enabled functions in EEPROM.cpp with the native ones of the rp2040
uint32_t status = save_and_disable_interrupts();
restore_interrupts(status);
These functions are located in hardware/sync.h of the pico sdk
Then It built without an included Arduino.h in any of the source files needed for the lib.
```
bool EEPROMClass::commit() {
    if (!_size) {
        return false;
    }
    if (!_dirty) {
        return true;
    }
    if (!_data) {
        return false;
    }

    uint32_t status = save_and_disable_interrupts();
    rp2040.idleOtherCore();
    flash_range_erase((intptr_t)_sector - (intptr_t)XIP_BASE, 4096);
    flash_range_program((intptr_t)_sector - (intptr_t)XIP_BASE, _data, _size);
    rp2040.resumeOtherCore();
    restore_interrupts(status);

    return true;
}
```

Wiz-IO, 2021-05-29:
lib EEEPROM exist in this port
https://github.com/Wiz-IO/framework-wizio-pico/tree/main/arduino/libraries/RP2040/EEPROM

this ._EEPROM_start = . is without dot at the beginning
**_EEPROM_start = .;**

goto .....platformio\packages\framework-wizio-pico\SDK\pico\pico_standard_link
copy / paste memmap_default.ld to eeprom.ld

```
    .flash_end : {
        . = ALIGN(4096); /* align to page*/
        _EEPROM_start = .;
        . = . + 4096; /* eeprom size in bytes */       
        __flash_binary_end = .;
    } > FLASH
```

add to platformio.ini
```
board_build.linker = eeprom.ld ; select other linker script
```

```
#include <Arduino.h>
#include <EEPROM.h>

extern "C" char _EEPROM_start;
extern "C" char __flash_binary_end;

void setup()
{
  Serial.begin(115200, true);
  printf("\n[APP] EEPROM 2021 Georgi Angelov\n");

  printf("EEPROM Start = %p\n", &_EEPROM_start);
  printf("EEPROM End   = %p\n", &__flash_binary_end);
  printf("EEPROM Size  = %u\n", &__flash_binary_end - &_EEPROM_start);

  EEPROM.begin(512);
}

void loop()
{
}

/*
  [APP] EEPROM 2021 Georgi Angelov
  EEPROM Start = 0x10009000
  EEPROM End   = 0x1000a000
  EEPROM Size  = 4096
*/
```

fluffymadness, 2021-05-29:
Thank you, i tested it and it is working.

closed 2021-05-29 by fluffymadness