Wiz-IO/wizio-pico #77: tinyusb debug over uart stopped working

fluffymadness, 2021-06-14:
On the newest sdk version i've lost the debug output for tinyusb when building with the
#define CFG_TUSB_DEBUG 2 for debug output.

I just had to connect pin0 and pin1 with rx and tx of a usb serial adapter and
I got debug out via serial.

I tested this side by side. Both tests where done on the same windows machine.

Flashing my microcontroller via windows where i have platform io installed with the previous version of wizio-pico
= debug output via uart

Flashing my microcontroller via linux where i have platformio with newest wizio-pico
= debug output gone via uart

I have a main.c file for testing.
Debug output was also not working for me when using a main.cpp file on the older wizio-pico
only via main.c

I played around a few hours but I sadly still haven't a solution
```
[env:raspberry-pi-pico]
platform = wizio-pico
board = raspberry-pi-pico
framework = baremetal ; 
 
monitor_port = SERIAL_PORT
monitor_speed = 115200

build_flags = 
    -D PICO_USB   ; both
    -D PICO_STDIO_UART  ; only baremetal

board_build.linker = eeprom.ld ; select other linker script
```

Wiz-IO, 2021-06-14:
Hi
this -D PICO_STDIO_UART in iNI file must  redirect printf to default uart
for debug tinyusb use simple printf
https://github.com/Wiz-IO/framework-wizio-pico/blob/master/SDK112/lib/tinyusb/src/common/tusb_common.h#L222

I didn't understand you....

Wiz-IO, 2021-06-14:
:) thanks Raspbery
the new key is
-D **LIB_PICO_STDIO_UART**

Wiz-IO, 2021-06-14:
btw: update platform - I make fix for LIB_PICO_STDIO_... blah :)

fluffymadness, 2021-06-15:
thanks :) just tested it. working as expected now.

closed 2021-06-15 by fluffymadness