Wiz-IO/wizio-pico #79: Upload function treats the monitor_port glob expression as an exact value

zapta, 2021-06-15:
The value of the platformio monitor_port flag is a globexpression but the Pic uploader treats it as a specific port name and thus fails to upload when the value is a glob.

https://docs.platformio.org/en/latest/projectconf/section_env_monitor.html#monitor-port

To reproduce:
* Let's say your port is COM3.
* Define in platformio.ini  monitor_port=COM[234]
* Try to build and upload
* Open the platformio serial monitor.

Expected behavior: Both upload and serial monitor work.
Actual behavior: Upload fails (does not find the port) but serial monitor does find it.

Wiz-IO, 2021-06-16:
platform not use upload_port

zapta, 2021-06-16:
It uses monitor_port here. Per the documentation of platformio, the value of monitor_port is a  regular expression (glob), not a port name.

https://github.com/Wiz-IO/wizio-pico/blob/main/builder/frameworks/uf2conv.py#L325

I think the right logic is to iterate the available ports and check which one matches the glob expression.  I think this is what the platformio serial monitor does since it is able to find the correct port.

Wiz-IO, 2021-06-16:
this is used for USB RESET

zapta, 2021-06-16:
Yes, this is how it forces the Pico to enter boot mode so it can upload to the disk drive.  But, if it think that the regex is a port name, it doesn't find the port, can't reset the board, and fail to upload ;-)

The main point is that monitor_port is not a port name but a regular expression.

Wiz-IO, 2021-06-16:
ignore upload_port
for monitor_port
if "com" is one PIO will add this port if monitor_port is ignored
else need manual select

zapta, 2021-06-16:
I don't use upload_port and never mentioned it here. I just asked about monitor_port. ;-)  It's a glob expression so I think the code should treat it like that, matching it against available ports.

Let's see what platformio has to say, maybe they have already code that does that. I posted a question here

https://community.platformio.org/t/builder-get-monitor-port-value-from-ini/20425/7

Meanwhile I use monitor_port=<exact-port-name-no-glob>

Wiz-IO, 2021-06-16:
sorry, I did not understand you

monitor_port = COMx - manual select, if there are more, I can't select - this is a function of PIO

zapta, 2021-06-16:
Please see the answer of @maxgerhart at the bottom of this thread

https://community.platformio.org/t/builder-get-monitor-port-value-from-ini/20425/8

If monitor_port is a glob expression, you can find the port to use by
matching the list of existing ports to the glob expression.

(I know that only one port is used, but since monitor_port in platformio
can be a glob expression, finding the port require port matching).

Wiz-IO, 2021-06-16:
I don't think it makes sense to complicate things...

zapta, 2021-06-17:
Sounds reasonable.

I added to the sample project pull request a clarification that
monitor_port should be an exact port name, globs not allowed.

https://github.com/Wiz-IO/wizio-pico/pull/78

Wiz-IO, 2021-06-17:
https://github.com/Wiz-IO/wizio-pico-examples

zapta, 2021-06-17:
Ok, thanks. I closed the pull request.

closed 2021-08-28 by Wiz-IO