Wiz-IO/wizio-pico #80: C++ 17 build doesn't work with baremetal

fluffymadness, 2021-07-02:
I don't know if it is possible but i tried to build with c++ 17 via buildflags.
Maybe i did something wrong,
however even when using baremetal, arduino dependencies seemed to be pulled in.
Namely the wiring lib. Is it possible to build without wiring ?
```
[env:raspberry-pi-pico]
platform = wizio-pico
board = raspberry-pi-pico
framework = baremetal ; 
 
monitor_port = SERIAL_PORT
monitor_speed = 115200

build_flags = 
    -D PICO_USB   ; both
    -D PICO_STDIO_UART  ; only baremetal
    -std=c++17

board_build.linker = eeprom.ld ; select other linker script
```

```
These are the kind of errors I get

Compiling .pio\build\raspberry-pi-pico\lib5dd\wiring\wiring_digital.o
In file included from C:\Users\chainer\.platformio\packages\framework-wizio-pico\library\wiring\include/Arduino.h:40:0,
                 from C:\Users\chainer\.platformio\packages\framework-wizio-pico\library\wiring\src\Wire.cpp:19:
C:\Users\chainer\.platformio\packages\framework-wizio-pico\library\wiring\include/WCharacter.h: In function 'bool arduino::isAscii(int)':
C:\Users\chainer\.platformio\packages\framework-wizio-pico\library\wiring\include/WCharacter.h:66:12: error: 'isascii' was not declared in this scope
   return ( isascii (c) == 0 ? false : true);
            ^~~~~~~
compilation terminated due to -Wfatal-errors.
*** [.pio\build\raspberry-pi-pico\lib5dd\wiring\Wire.o] Error 1
In file included from C:\Users\chainer\.platformio\packages\framework-wizio-pico\library\wiring\include/Arduino.h:40:0,
```

fluffymadness, 2021-07-02:
ok this seems to have fixed it

build_flags = 
    -D PICO_USB   ; both
    -D PICO_STDIO_UART  ; only baremetal
    -std=c++17
    -std=gnu++17
build_unflags =
    -std=gnu++11

fluffymadness, 2021-07-02:
i still have an error, i have commented_out the to_ascii function
for it to build

/*
// Converts c to a 7-bit unsigned char value that fits into the 
// ASCII character set, by clearing the high-order bits.
inline int toAscii(int c)
{
  return toascii (c);
}*/

Its in the wiring library in WCharacter.h

Wiz-IO, 2021-07-02:
hi
for other C++X
you need implement missing functions

closed 2021-08-28 by Wiz-IO