Wiz-IO/wizio-pico #83: Upload error

MisterSandman7, 2021-10-12:
Upload error, even with path specified
On Manjaro 21.1.5 
```pio run --target upload --upload-port=/run/media/[USER]/RPI-RP2 --verbose
Processing raspberry-pi-pico (platform: wizio-pico; board: pico-dap; framework: baremetal; upload_port: /run/media/[USER]/RPI-RP2; board_build.heap: 4096; board_build.stack: 4096; board_build.boot: w25q080; board_build.bynary_type: copy_to_ram; build_flags: -D LIB_PICO_STDIO_USB; build_unflags: -D LIB_PICO_STDIO_UART)
------------------------------------------------------------------------------------------------------------------------------------------------------------------
<<<<<<<<<<<< WIZIO - RASPBERRY PI PICO ( CMSIS-DAP ) 2021 Georgi Angelov >>>>>>>>>>>>
CONFIGURATION: https://docs.platformio.org/page/boards/wizio-pico/pico-dap.html
PLATFORM: WizIO - Raspberry Pi Pico (1.0.9+sha.abeb4e1) (git+https://github.com/Wiz-IO/wizio-pico) > WizIO - Raspberry Pi Pico ( CMSIS-DAP )
HARDWARE: RP2040 125MHz, 256KB RAM, 2MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, picoprobe)
PACKAGES: 
 - framework-wizio-pico 1.0.9+sha.6205459 (git+https://github.com/Wiz-IO/framework-wizio-pico.git) 
 - tool-wizio-pico 1.0.0+sha.3a504e1 (git+https://github.com/Wiz-IO/tool-wizio-pico.git) 
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)

SDK RASPBERRYPI PI PICO RP2040 ( PICO - SDK )
  * OPTIMIZATION : -Os
  * STACK        : 4096
  * HEAP         : 4096
  * SPECS        : nano.specs
  * BINARY TYPE  : copy_to_ram ['w25q080', 'memmap_copy_to_ram.ld', '0x10000000']
  * * STDIO USB
  * TINYUSB

LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 9 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
MethodWrapper(["checkprogsize"], [".pio/build/raspberry-pi-pico/APPLICATION.elf"])
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   2.5% (used 6556 bytes from 262144 bytes)
Flash: [          ]   2.0% (used 42788 bytes from 2097152 bytes)
.pio/build/raspberry-pi-pico/APPLICATION.elf  :
section                              size        addr
.boot2                                256   268435456
.flashtext                            476   268435712
.rodata                                 0   268436188
.ARM.exidx                              8   268436188
.binary_info                           24   268436196
.ram_vector_table                     192   536870912
.text                               39696   536871104
.data                                2836   536910800
.uninitialized_ram.magic_location      12   536913636
.uninitialized_data                     0   536913648
.scratch_x                              0   537133056
.scratch_y                              0   537137152
.bss                                 3528   536913648
.heap                                4096   536917176
.stack_dummy                         4096   537137152
.comment                              126           0
.ARM.attributes                        48           0
.debug_frame                         4680           0
Total                               60074
<lambda>(["upload"], ["buildprog"])
AVAILABLE: cmsis-dap, picoprobe, uf2
CURRENT: upload_protocol = uf2
dev_uploader(["upload"], ["buildprog"])
  Converting to UF2 ( 0x10000000 )
  Wrote 87040 bytes to /home/[USER]/Documents/PlatformIO/Projects/Pico/blink/.pio/build/raspberry-pi-pico/APPLICATION.uf2
*** [upload] /media: No such file or directory```

Wiz-IO, 2021-10-13:
I not have Manjaro ... cant test
Try edit
https://github.com/Wiz-IO/wizio-pico/blob/main/builder/frameworks/uf2conv.py#L317

MisterSandman7, 2021-10-13:
tried some print debug and found the upload fails at line 338 of `uf2cov.py`, at the line `drives = get_drives()`, not sure why yet

MisterSandman7, 2021-10-13:
Been experimenting and found it fails at line 337 of `uf2conv.py`, line `drives = get_drives()`, not sure why yet

Wiz-IO, 2021-10-13:
print( sys.platform ) 
and add path to rootpath = ???

Wiz-IO, 2021-10-13:
btw uploader is tested on MAC, LIN and WIN

Wiz-IO, 2021-10-13:
tyr add '/run'

tmp = join( 'run', rootpath,  os.environ["USER"] )

nikisalli, 2021-11-11:
is this going to be fixed upstream? @MisterSandman7's solution works fine

closed 2021-11-25 by Wiz-IO

Project-FreeMind, 2021-12-07:
Normally, you have in Linux a `media` folder under root `/`. I don't know, why arch does not have this, but you can create a symlink `sudo ln -s /run/media /media`. After that, upload should work as expected.

Regards