Wiz-IO/wizio-pico #85: undefined reference to `getchar_timeout_us'

rishubil, 2021-12-20:
I can't build the following code.
Is there anything I'm missing?

main.cpp
```cpp
#include <Arduino.h>
#include <pico/stdio.h>

void setup() {
  char c  = getchar_timeout_us((uint32_t)0);
}

void loop() {
}
```

platformio.ini
```ini
[env:pico-dap]
platform = wizio-pico
board = pico-dap
framework = arduino

monitor_port = COM5
monitor_speed = 115200

build_flags =
    -D PICO_STDIO_USB
```

Build log
```
> Executing task: C:\Users\username\.platformio\penv\Scripts\platformio.exe run --verbose --environment pico-dap <

Processing pico-dap (platform: wizio-pico; board: pico-dap; framework: arduino; monitor_port: COM5; monitor_speed: 115200; build_flags: -D PICO_STDIO_USB)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------<<<<<<<<<<<< WIZIO - RASPBERRY PI PICO ( CMSIS-DAP ) 2021 Georgi Angelov >>>>>>>>>>>>
CONFIGURATION: https://docs.platformio.org/page/boards/wizio-pico/pico-dap.html
PLATFORM: WizIO - Raspberry Pi Pico (1.0.9+sha.abeb4e1) (git+https://github.com/Wiz-IO/wizio-pico) > WizIO - Raspberry Pi Pico ( CMSIS-DAP )
HARDWARE: RP2040 125MHz, 256KB RAM, 2MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, picoprobe)
PACKAGES:
 - framework-wizio-pico 1.0.9+sha.6205459 (git+https://github.com/Wiz-IO/framework-wizio-pico.git)
 - tool-wizio-pico 1.0.0+sha.3a504e1 (git+https://github.com/Wiz-IO/tool-wizio-pico.git)
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)

ARDUINO RASPBERRYPI PI PICO RP2040 ( PICO - SDK )
  * OPTIMIZATION : -Os
  * STACK        : 2048
  * HEAP         : maximum
  * SPECS        : nano.specs
  * BINARY TYPE  : default ['w25q080', 'memmap_default.ld', '0x10000000']
  * TINYUSB

LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Framework incompatible library C:\Users\username\.platformio\packages\framework-wizio-pico\library\wiring
Found 17 compatible libraries
More details about "Library Compatibility Mode": https://docs.platformio.org/page/librarymanager/ldf.html#ldf-compat-mode
Scanning dependencies...
No dependencies
Building in release mode
arm-none-eabi-g++ -o .pio\build\pico-dap\src\main.cpp.o -c -fno-rtti -fno-exceptions -fno-threadsafe-statics -fno-non-call-exceptions -fno-use-cxa-atexit -march=armv6-m -mcpu=cortex-m0plus -mthumb -Os -fdata-sections -ffunction-sections -Wall -Wextra -Wfatal-errors -Wno-sign-compare -Wno-type-limits -Wno-unused-parameter -Wno-unused-function -Wno-unused-but-set-variable -Wno-unused-variable -Wno-unused-value -Wno-strict-aliasing -Wno-maybe-uninitialized -DPLATFORMIO=50205 -DPICO_STDIO_USB -DLIB_PICO_STDIO_USB -DPICO_STDIO_USB -DNDEBUG -DPICO_ON_DEVICE=1 -DPICO_HEAP_SIZE=2048 -DPICO_STACK_SIZE=2048 -DARDUINO=200 -DCFG_TUSB_MCU=OPT_MCU_RP2040 -DCFG_TUSB_OS=OPT_OS_PICO -Iinclude -Isrc -Isrc -Ilib -Iinclude -IC:\Users\username\.platformio\packages\framework-wizio-pico\wizio\pico -IC:\Users\username\.platformio\packages\framework-wizio-pico\wizio\newlib -IC:\Users\username\.platformio\packages\framework-wizio-pico\SDK\include -IC:\Users\username\.platformio\packages\framework-wizio-pico\SDK\cmsis\include -IC:\Users\username\.platformio\packages\framework-wizio-pico\arduino\arduino -IC:\Users\username\.platformio\packages\framework-wizio-pico\arduino\cores\RP2040 -IC:\Users\username\.platformio\packages\framework-wizio-pico\arduino\variants\raspberry-pi-pico -IC:\Users\username\.platformio\packages\framework-wizio-pico\SDK\lib\tinyusb\src src\main.cpp
arm-none-eabi-g++ -o .pio\build\pico-dap\ARDUINO.elf -T C:\Users\username\.platformio\packages\framework-wizio-pico\SDK\pico\pico_standard_link\memmap_default.ld -march=armv6-m -mcpu=cortex-m0plus -mthumb -Os -nostartfiles -Xlinker --gc-sections -Wl,--gc-sections --entry=_entry_point -specs=nano.specs -u _printf_float -u _scanf_float .pio\build\pico-dap\arduino\arduino\arduino\IPAddress.cpp.o .pio\build\pico-dap\arduino\arduino\arduino\Print.cpp.o .pio\build\pico-dap\arduino\arduino\arduino\Stream.cpp.o .pio\build\pico-dap\arduino\arduino\arduino\WMath.cpp.o .pio\build\pico-dap\arduino\arduino\arduino\WString.cpp.o .pio\build\pico-dap\arduino\arduino\arduino\cbuf.cpp.o .pio\build\pico-dap\arduino\arduino\arduino\dtostrf.c.o .pio\build\pico-dap\arduino\arduino\arduino\new.cpp.o .pio\build\pico-dap\arduino\arduino\arduino\wiring_pulse.c.o .pio\build\pico-dap\arduino\arduino\arduino\wiring_shift.c.o .pio\build\pico-dap\arduino\arduino\core\Tone.cpp.o .pio\build\pico-dap\arduino\arduino\core\arduino_debug.c.o .pio\build\pico-dap\arduino\arduino\core\arduino_heap_4.c.o .pio\build\pico-dap\arduino\arduino\core\arduino_main.cpp.o .pio\build\pico-dap\arduino\arduino\core\arduino_stdio_usb.c.o .pio\build\pico-dap\arduino\arduino\core\arduino_stdio_usb_descriptors.c.o .pio\build\pico-dap\arduino\arduino\core\interface.c.o .pio\build\pico-dap\arduino\arduino\core\wiring.c.o .pio\build\pico-dap\arduino\arduino\variant\pins.c.o .pio\build\pico-dap\arduino\arduino\variant\variant.cpp.o .pio\build\pico-dap\arduino\wizio\newlib\lock.c.o .pio\build\pico-dap\arduino\wizio\newlib\syscall.c.o .pio\build\pico-dap\arduino\wizio\newlib\time.c.o .pio\build\pico-dap\arduino\wizio\pico\crt0.S.o .pio\build\pico-dap\arduino\wizio\pico\memory.c.o .pio\build\pico-dap\arduino\wizio\pico\wizio.c.o .pio\build\pico-dap\arduino\wizio\boot\boot2.S.o .pio\build\pico-dap\arduino\SDK\pico\usb\common\tusb_fifo.c.o .pio\build\pico-dap\arduino\SDK\pico\usb\portable\raspberrypi\rp2040\dcd_rp2040.c.o .pio\build\pico-dap\arduino\SDK\pico\usb\portable\raspberrypi\rp2040\hcd_rp2040.c.o .pio\build\pico-dap\arduino\SDK\pico\usb\portable\raspberrypi\rp2040\rp2040_usb.c.o .pio\build\pico-dap\arduino\SDK\pico\usb\tusb.c.o .pio\build\pico-dap\arduino\SDK\pico\usb\class\cdc\cdc_device.c.o .pio\build\pico-dap\arduino\SDK\pico\usb\class\cdc\cdc_host.c.o .pio\build\pico-dap\arduino\SDK\pico\usb\class\cdc\cdc_rndis_host.c.o .pio\build\pico-dap\arduino\SDK\pico\usb\device\usbd.c.o .pio\build\pico-dap\arduino\SDK\pico\usb\device\usbd_control.c.o .pio\build\pico-dap\arduino\SDK\cmsis\src\system_RP2040.c.o .pio\build\pico-dap\arduino\SDK\hardware\adc.c.o .pio\build\pico-dap\arduino\SDK\hardware\claim.c.o .pio\build\pico-dap\arduino\SDK\hardware\clocks.c.o .pio\build\pico-dap\arduino\SDK\hardware\divider.S.o .pio\build\pico-dap\arduino\SDK\hardware\dma.c.o .pio\build\pico-dap\arduino\SDK\hardware\exception.c.o .pio\build\pico-dap\arduino\SDK\hardware\flash.c.o .pio\build\pico-dap\arduino\SDK\hardware\gpio.c.o .pio\build\pico-dap\arduino\SDK\hardware\i2c.c.o .pio\build\pico-dap\arduino\SDK\hardware\interp.c.o .pio\build\pico-dap\arduino\SDK\hardware\irq.c.o .pio\build\pico-dap\arduino\SDK\hardware\irq_handler_chain.S.o .pio\build\pico-dap\arduino\SDK\hardware\pio.c.o .pio\build\pico-dap\arduino\SDK\hardware\pll.c.o .pio\build\pico-dap\arduino\SDK\hardware\rtc.c.o .pio\build\pico-dap\arduino\SDK\hardware\spi.c.o .pio\build\pico-dap\arduino\SDK\hardware\sync.c.o .pio\build\pico-dap\arduino\SDK\hardware\timer.c.o .pio\build\pico-dap\arduino\SDK\hardware\uart.c.o .pio\build\pico-dap\arduino\SDK\hardware\vreg.c.o .pio\build\pico-dap\arduino\SDK\hardware\watchdog.c.o .pio\build\pico-dap\arduino\SDK\hardware\xosc.c.o .pio\build\pico-dap\arduino\SDK\include\hardware\divider_helper.S.o .pio\build\pico-dap\arduino\SDK\include\pico\asm_helper.S.o .pio\build\pico-dap\arduino\SDK\pico\pico_bootrom\bootrom.c.o .pio\build\pico-dap\arduino\SDK\pico\pico_bootsel_via_double_reset\pico_bootsel_via_double_reset.c.o .pio\build\pico-dap\arduino\SDK\pico\pico_fix\rp2040_usb_device_enumeration\rp2040_usb_device_enumeration.c.o .pio\build\pico-dap\arduino\SDK\pico\pico_multicore\multicore.c.o .pio\build\pico-dap\arduino\SDK\pico\pico_platform\platform.c.o .pio\build\pico-dap\arduino\SDK\pico\pico_runtime\runtime.c.o .pio\build\pico-dap\arduino\SDK\pico\pico_standard_link\binary_info.c.o .pio\build\pico-dap\arduino\SDK\pico\pico_stdlib\stdlib.c.o .pio\build\pico-dap\arduino\SDK\pico\pico_sync\critical_section.c.o .pio\build\pico-dap\arduino\SDK\pico\pico_sync\lock_core.c.o .pio\build\pico-dap\arduino\SDK\pico\pico_sync\mutex.c.o .pio\build\pico-dap\arduino\SDK\pico\pico_sync\sem.c.o .pio\build\pico-dap\arduino\SDK\pico\pico_time\time.c.o .pio\build\pico-dap\arduino\SDK\pico\pico_time\timeout_helper.c.o .pio\build\pico-dap\arduino\SDK\pico\pico_unique_id\unique_id.c.o .pio\build\pico-dap\arduino\SDK\pico\pico_util\datetime.c.o .pio\build\pico [... cut ...]
main.cpp:(.text._Z5setupv+0x4): undefined reference to `getchar_timeout_us'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\pico-dap\ARDUINO.elf] Error 1
======================================================================= [FAILED] Took 2.29 seconds =======================================================================
The terminal process "C:\Users\username\.platformio\penv\Scripts\platformio.exe 'run', '--verbose', '--environment', 'pico-dap'" terminated with exit code: 1.
```

Wiz-IO, 2021-12-20:
Arduino not support direct call to stdio
**getchar_timeout_us()** is C function... but you call from CPP

rishubil, 2021-12-20:
I see. I'll try something else. Thank you.

closed 2021-12-20 by rishubil