Wiz-IO/wizio-pico #94: "Upload only" using Picoprobe

smoosun, 2022-04-22:
Hello, I am on Windows 10.

I can use my picoprobe to debug without any problem. It works perfectly !

However, when I only want to upload my code (without wanting to debug it) by clicking on the "PlatformIO: Upload" arrow, it tells me that it cannot find the Pico USB drive :

Configuring upload protocol...
AVAILABLE: cmsis-dap, picoprobe, uf2
CURRENT: upload_protocol = picoprobe
Uploading...
  Converting to UF2 ( 0x10000000 )
  Wrote 69632 bytes to C:/Users/xxxxx/Documents/PlatformIO/Projects/TestWizio2/.pio/build/raspberry-pi-pico\APPLICATION.uf2

   Pico USB drive not found.

Here is my platformio.ini :

[env:raspberry-pi-pico]
platform = wizio-pico
board = raspberry-pi-pico
framework = baremetal
upload_protocol = picoprobe
debug_tool = picoprobe

Is it a bug ? Did I forget something ? Thanks !

Wiz-IO, 2022-04-22:
Hi
remove upload_protocol = picoprobe
in release mode uploader use uf2
in debug mode picoprobe will upload debug.elf to pico

smoosun, 2022-04-22:
Thank you for your answer !

I thought it was possible to upload the program without launching the debugger. I find it faster to do some tests.

When I was using the Pico build chain I had a script that uploaded the program via OpenOCD when I clicked on the green arrow in VSCode. I'll see if I can use it. Maybe I have some bad habits? I find it slower to go through the debugger; maybe I'm using it wrong?

Thanks !

closed 2022-04-22 by smoosun

dlbogdan, 2022-04-22:
There are several reasons why you’d want to upload through swd (with picoprobe) even for a release. For example when you implement usb-host on the pico. It’’s too much stress for the microusb header while testing.

smoosun, 2022-04-22:
With my previous environment (before Wiz-IO) I used a configuration in launch.json :

{
            "name": "Cortex Run",
            "cwd": "${workspaceRoot}",
            "executable": "${workspaceRoot}/build/blink.elf",
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "openocd",
            "device": "Pico2040",
            //"runToMain": true,
            "stopOnEntry": false,
            "stopAtEntry": false,
            "configFiles": ["interface/picoprobe.cfg", "target/rp2040.cfg"],
            "searchDir": ["C:\\VSARM\\openocd\\tcl"],
            "svdFile": "C:\\VSARM\\sdk\\pico\\pico-sdk\\src\\rp2040\\hardware_regs\\rp2040.svd",
}

Currently it doesn't work well, Wiz-IO regenerates the launch.json file automatically.

I liked to be able to launch a "release" (without launching the debugger) it was fast (the code is compact, quick to transfer) and convenient (no need to handle the cable and the push button).

Wiz-IO, 2022-04-23:
**launch.json** is created from PlatformIO core, not from me