Wiz-IO/wizio-pico #96: USB console does not work by default

zshivers, 2022-04-29:
When you create a new project, the example code includes using `printf()` but the build_flags aren't configured by default to give you the USB stdio. For the code to work, I had to go add `build_flags = -DLIB_PICO_STDIO_USB` into platform.ini like the wiki says. Maybe this should be the default config?

zshivers, 2022-04-29:
This is the code automatically populated into `main.c` when creating a new project using PlatformIO using the Baremetal setting:
```
#include <stdio.h>
#include "pico/stdlib.h"

int main(void)
{  
    stdio_init_all();
    printf("\n\nHello World\n");
    gpio_init(PICO_DEFAULT_LED_PIN);
    gpio_set_dir(PICO_DEFAULT_LED_PIN, GPIO_OUT);
    while (true)
    {
        gpio_put(PICO_DEFAULT_LED_PIN, 1);
        sleep_ms(100);
        gpio_put(PICO_DEFAULT_LED_PIN, 0);
        sleep_ms(100);
    }
}
```

Wiz-IO, 2022-04-29:
yes, this is default template and printf()  ( or delete line ) is not active without flag LIB_PICO_STDIO_xxx
so it was in the original SDK ... now i don't know how it is
I am not working on this project due to incorrect behavior on the part of the Raspberry PI org

closed 2022-04-29 by Wiz-IO