Wiz-IO/wizio-pico #97: UnicodeDecodeError

3dzavr, 2022-05-13:
When I installed and tried to compile, I got an error:
[image]

Wiz-IO, 2022-05-13:
Hi
uf2conv.py is Microsoft code
https://github.com/microsoft/uf2/blob/master/utils/uf2conv.py

it is very likely a Cyrillic in PATH name

3dzavr, 2022-05-13:
Path to project : c:\Users\AlexSVC\Documents\PlatformIO\Projects\Pico_test\
No Cyrillic ...
I moved the folder to the root, to another drive, but that didn't change anything.
Where else to look for the cause?

Wiz-IO, 2022-05-13:
https://github.com/Wiz-IO/wizio-pico/blob/main/builder/frameworks/uf2conv.py#L200

add after line: 
print( r ), look where is this utf-8 simbol 0x8D

Wiz-IO, 2022-05-13:
I can't repeat the bug to debug line

3dzavr, 2022-05-13:
b'DeviceID  DriveType  FileSystem  VolumeName          \r\r\nA:        3          NTFS        \x8d\xae\xa2\xeb\xa9 \xe2\xae\xac   \r\r\nB:
      3          NTFS        \x8d\xae\xa2\xeb\xa9 \xe2\xae\xac   \r\r\nC:        3          NTFS                            \r\r\nD:
3          NTFS        \x95\xe0\xa0\xad\xa8\xab\xa8\xe9\xa5  \r\r\nE:        3          NTFS        SSD                 \r\r\n\r\r\n'

3dzavr, 2022-05-13:
Could it be the volume label?

3dzavr, 2022-05-13:
Removed the Cyrillic alphabet in the volume label, now:
"Pico USB drive not found."
:)

Wiz-IO, 2022-05-13:
уже работает ?

3dzavr, 2022-05-13:
Нет, говорит "Pico USB drive not found."

3dzavr, 2022-05-13:
File platfomio.ini:
[env:raspberry-pi-pico]
platform = wizio-pico
board = raspberry-pi-pico
framework = arduino

upload_protocol = picoprobe
debug_tool = picoprobe

Wiz-IO, 2022-05-13:
[image]

Wiz-IO, 2022-05-13:
[image]

3dzavr, 2022-05-13:
Я пытаюсь запустить связку двух пико. Одна пикопроба, вторая как цель. Пикопроба прошил picoprobe.uf2, драйвер установил zadig-2.7.exe.

3dzavr, 2022-05-13:
Ожидалось, что прошивка будет через picoprobe. По факту получилось пока войти в отладку. Просто прошить пока нет.
[image]

Wiz-IO, 2022-05-13:
for debug not need upload - just click Start Debugung - F5

Wiz-IO, 2022-05-13:
https://www.youtube.com/watch?v=SdpsmgRp5Co

3dzavr, 2022-05-13:
Debug working. I just wanted to flash without entering the debug mode. But the upload button gives an error "Pico USB drive not found".
[image]
I don't have a General tab. How to turn it on?

Wiz-IO, 2022-05-13:
pico buton hold, connect USB, upload
b'DeviceID  DriveType  FileSystem  VolumeName  \r\r\nC:        3          NTFS                    \r\r\nD:        3          NTFS        PROJECTS    \r\r\nE:        5                                  \r\r\nF:        3          NTFS
      STORAGE     \r\r\nG:        2          FAT         **RPI-RP2**     \r\r\n\r\r\n'

3dzavr, 2022-05-13:
:) 
Okay, so far I'll do it. Of course, I wanted to download it by pressing the button, but I would have to rearrange the USB cable.
Thanks for the help! If there is an opportunity and desire, complete the ability to download by button. I will be very grateful!