Wiz-IO/wizio-pico #98: Cannot import name 'dev_pioasm' from 'pioasm'

oomek, 2022-05-17:
I installed your platform, downloaded some examples, but I keep getting these errors, any clues what might be causing it?

```
> Executing task: C:\Users\Radek\.platformio\penv\Scripts\platformio.exe run <

Processing raspberry-pi-pico (platform: wizio-pico; board: raspberry-pi-pico; framework: baremetal)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
ImportError: cannot import name 'dev_pioasm' from 'pioasm' (C:\Users\Radek\.platformio\penv\lib\site-packages\platformio\builder\tools\pioasm.py):
  File "C:\Users\Radek\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 184:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\Radek\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Script\SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Radek\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\Radek\.platformio\platforms\wizio-pico\builder\main.py", line 9:
    from pioasm import dev_pioasm
===================================================================================== [FAILED] Took 0.61 seconds =====================================================================================
The terminal process "C:\Users\Radek\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
```

Wiz-IO, 2022-05-17:
**pioasm.py** exist in platform
https://github.com/Wiz-IO/wizio-pico/tree/main/builder

or delete folder 
C:\Users\Radek\.platformio\platforms\wizio-pico
and install again

oomek, 2022-05-17:
I've already tried to delete all the wiz folders and reinstalling. These files are there, just for some reason pio is looking for them in the wrong folder.

I have to mention one thing, the installation failed repeatedly in the beginning with git errors until I disabled exploit protection in Windows 11. It's a known issue. Maybe some config files got corrupted, I have no clue what is going on.

oomek, 2022-05-17:
Tried to uninstall again and now I'm getting this during installation:

```
Could not install platform
PIO Core Call Error: "Platform Manager: Installing git+https://github.com/Wiz-IO/wizio-pico\r\ngit version 2.36.1.windows.1\n\n\nCloning into 'C:\\Users\\Radek\\.platformio\\.cache\\tmp\\pkg-installing-n31htnu0'...\nC:/Program Files/Git/mingw64/libexec/git-core\\git-submodule: line 637: cmd_: command not found\nError: VCS: Could not process command ['git', 'clone', '--recursive', '--depth', '1', 'https://github.com/Wiz-IO/wizio-pico', 'C:\\\\Users\\\\Radek\\\\.platformio\\\\.cache\\\\tmp\\\\pkg-installing-n31htnu0']"
```

oomek, 2022-05-17:
After disabling globally ASLR in Windows Exploit Protection I was able to install again
```
Platform Manager: Installing git+https://github.com/Wiz-IO/wizio-pico
git version 2.36.1.windows.1
Platform Manager: wizio-pico@1.0.9+sha.abeb4e1 has been installed!
Tool Manager: Installing git+https://github.com/Wiz-IO/framework-wizio-pico.git
git version 2.36.1.windows.1
Tool Manager: framework-wizio-pico@1.0.9+sha.6205459 has been installed!
Tool Manager: Installing platformio/toolchain-gccarmnoneeabi @ ~1.70201.0
Downloading
Unpacking
Tool Manager: toolchain-gccarmnoneeabi@1.70201.0 has been installed!
Tool Manager: Installing git+https://github.com/Wiz-IO/tool-wizio-pico.git
git version 2.36.1.windows.1
Tool Manager: tool-wizio-pico@1.0.0+sha.3a504e1 has been installed!
The platform 'https://github.com/Wiz-IO/wizio-pico' has been successfully installed!
The rest of the packages will be installed later depending on your build environment.
Cloning into 'C:\Users\Radek\.platformio\.cache\tmp\pkg-installing-y49bx3s6'...
Cloning into 'C:\Users\Radek\.platformio\.cache\tmp\pkg-installing-5__wqj_f'...
Cloning into 'C:\Users\Radek\.platformio\.cache\tmp\pkg-installing-2wi0cy3b'...
```
Unfortunately the problem with building still persist

Wiz-IO, 2022-05-17:
try manual inastall - download:
https://github.com/Wiz-IO/wizio-pico
https://github.com/Wiz-IO/framework-wizio-pico

and unachive to platformIO folders
C:\Users\USER\.platformio\**platforms**\wizio-pico
C:\Users\USER\.platformio\**packages**\framework-wizio-pico

oomek, 2022-05-17:
I've just tried, The same error persist unfortunately.

Wiz-IO, 2022-05-17:
I have no idea, it works for me, I can't repeat the bug

3dzavr, 2022-05-17:
I have the same problem. Windows 7. Reinstall Plamtformio and Wiz-IO - The same error persist unfortunately.

3dzavr, 2022-05-17:
\.platformio\penv\Lib\site-packages\platformio\builder\tools\
There is pioasm.py, and it is called instead of the expected one.
There are two files and the wrong one is called.

3dzavr, 2022-05-17:
In main.py -- from pioasm import dev_pioasm -> from wpioasm import dev_pioasm
In the same folder -- pioasm.py -> rename to wpioasm.py
The error is gone.

oomek, 2022-05-17:
Ok but which main.py should I modify?

3dzavr, 2022-05-17:
[image]

oomek, 2022-05-17:
It worked! Thank you very much @3dzavr for the workaround.

3dzavr, 2022-05-17:
\.platformio\platforms\wizio-pico\builder\

3dzavr, 2022-05-17:
It's my pleasure :)

ivankravets, 2022-05-17:
@Wiz-IO please prefix `pioasm` module. It conflicts with the internal PlatformIO Core module https://github.com/platformio/platformio-core/blob/develop/platformio/builder/tools/pioasm.py

Wiz-IO, 2022-05-17:
done

closed 2022-05-17 by oomek