Wiz-IO/wizio-pico #101: I don't understand how to access to tinyusb functionality

Waaflee, 2022-06-15:
I'm trying to get this project to work with platformio  [pico_usb_kdb_mouse](https://github.com/a-h/pico_usb_kbd_mouse)
but I can't import tusb.h nor board.h. I've read the baremetal wiki up and down and I couldn't find anything referring to tinyusb lib setup.

Wiz-IO, 2022-06-16:
look this example
https://github.com/Wiz-IO/wizio-pico-examples/blob/main/baremetal/pico-usb-webserver/platformio.ini#L22

and edit INI for HID
https://github.com/a-h/pico_usb_kbd_mouse/blob/main/tusb_config.h#L3

Wiz-IO, 2022-06-16:
but usb host is not supported yet...
https://github.com/Wiz-IO/wizio-pico/blob/main/builder/frameworks/pico.py#L270

Waaflee, 2022-06-16:
Any hint on how could I include it in my fork? I've managed to compile but it doesn't seems to link it

```
Linking .pio/build/raspberry-pi-pico/APPLICATION.elf
.pio/build/raspberry-pi-pico/sdk/SDK/pico/usb/class/hid/hid_host.o: In function `hidh_init':
hid_host.c:(.text.hidh_init+0xc): undefined reference to `hidh_generic_init'
.pio/build/raspberry-pi-pico/sdk/SDK/pico/usb/class/hid/hid_host.o: In function `hidh_set_config':
hid_host.c:(.text.hidh_set_config+0x22): undefined reference to `tuh_hid_mouse_mounted_cb'
.pio/build/raspberry-pi-pico/sdk/SDK/pico/usb/class/hid/hid_host.o: In function `hidh_xfer_cb':
hid_host.c:(.text.hidh_xfer_cb+0x14): undefined reference to `tuh_hid_mouse_isr'
.pio/build/raspberry-pi-pico/sdk/SDK/pico/usb/class/hid/hid_host.o: In function `hidh_close':
hid_host.c:(.text.hidh_close+0x1c): undefined reference to `tuh_hid_mouse_unmounted_cb'
hid_host.c:(.text.hidh_close+0x22): undefined reference to `hidh_generic_close'
collect2: error: ld returned 1 exit status
*** [.pio/build/raspberry-pi-pico/APPLICATION.elf] Error 1
```

With this change
[image]

and this flag `-D PICO_USB`

Wiz-IO, 2022-06-17:
I will make update very soon

Wiz-IO, 2022-06-19:
now you can reinstall platform - support SDK **1.31** & 1.20
**tusb_config.h** is created from template to project/include for user editing

project INI
build_flags = -D PICO_USB_HOST load tinyusb as host
or as device
build_flags = -D PICO_USB_DEVICE and/or -D PICO_STDIO_USB ( for printf )
https://github.com/Wiz-IO/wizio-pico/blob/main/builder/frameworks/pico.py#L257

Wiz-IO, 2022-06-19:
example as beginning...
[PICO-HOST.zip](https://github.com/Wiz-IO/wizio-pico/files/8934889/PICO-HOST.zip)

Waaflee, 2022-06-22:
Will try!! Thanks a lot!!! I'll report how it went

Waaflee, 2022-06-22:
Well it compiles and apparently works. I've manage to print when the devices is mounted/unmounted but I seem to not be able to process any report or that they don't arrive. 
Can you give me a clue? hehe.
I cannot find any proper documentation in this regard.

Wiz-IO, 2022-06-23:
I not have experience tinyusb-host
Get some hub functions and find it in github - there is many examples

Wiz-IO, 2022-07-05:
Hi
are you test platform with SDK 1.3.1 ... I not have board for tests ( for now )
I'm interested: does the usb (device) work for debug as printf()

closed 2022-07-05 by Wiz-IO