Wiz-IO/wizio-pico #106: USB Serial not working with latest version

racerxdl, 2022-07-04:
I just did a quick hello world to test using the platform.io and used a hello world example in https://github.com/Wiz-IO/wizio-pico-examples/tree/main/HelloWorld/HELLO but a simpler version:

```c++
#include <Arduino.h>

void setup() {
  Serial.begin(115200, true);
  while(!Serial) {}
  Serial.println("TESTE");
  pinMode(25, OUTPUT);

}

void loop() {
    Serial.print("LO");
    Serial.println("OP");
    // Serial.printf("MILLIS = %u\n", millis());
    printf("PRINTF %f\n", 22.0 / 7);

    static int led = 0;
    digitalWrite(25, led);
    led ^= 1;
    delay(1000);
}
```

The led blinks fine, and the USB port is found (if arduino framework is selected) but nothing goes through the serial port (both with platform.io monitor and screen)

If I compile the same code using the default from platform.io everything works as expected.

My platformio.ini:

"Vanilla" platformio
```
[env:raspberry-pi-pico]
platform = raspberrypi
board = pico
framework = arduino

monitor_port = /dev/ttyACM0
monitor_speed = 115200

; build_flags =
;     -D PICO_STDIO_UART
;     -D PICO_STDIO_USB

; lib_deps = wiring

;lib_deps =
```

Wizio:

```
[env:raspberry-pi-pico]
platform = wizio-pico
board = raspberry-pi-pico
framework = arduino

monitor_port = /dev/ttyACM0
monitor_speed = 115200

; Also tried commenting out, and changing build flags
build_flags =
    -D PICO_STDIO_UART
    -D PICO_STDIO_USB

lib_deps = wiring

;lib_deps =
```

If I change the framework to baremetal, the led also blinks, but no ttyACM port appears (nothing on USB device list). I'm using ubuntu 22.04

Wiz-IO, 2022-07-05:
Hi
maybe some is changed in new SDK 1.3.1 ... I not have hardware to test now

but you use normal uart/Serial for printf()
for USB check this
https://github.com/Wiz-IO/framework-wizio-pico/tree/main/arduino/libraries/RP2040/SerialUSB

https://github.com/Wiz-IO/wizio-pico/wiki/ARDUINO#stdio-printf---only-one-driver-will-work

Wiz-IO, 2022-07-05:
try change SDK version to 1.2.0 ( is backup sdk for platform ) from ini file
this version has been working without problems for the last year

board_build.sdk  = SDK120 // witout this SDK version is 130

and test again...

Wiz-IO, 2022-07-05:
work...
`
#include <Arduino.h>

void setup()
{
  Serial.begin(115200); // Serial.begin(115200, true); will redirect printf() to this Serial [ uart ] 
  Serial.println("\nArdiuno Raspberrypi PI Pico 2021 Georgi Angelov");
  pinMode(LED, OUTPUT);
}

void loop()
{
  static int led = 0;
  digitalWrite(LED, led);
  led ^= 1;
  delay(1000);
  Serial.println( millis() ); // print to uart
  
  printf("loop"); // build_flags = -D PICO_STDIO_USB ... print to USB virtual port 
}
`

closed 2022-07-05 by Wiz-IO