Wiz-IO/wizio-pico #107: after last update problem with #include "cyw43.h"

Squarenix17, 2022-07-05:
[image]
Hi,
after the last update during the compilation of all projects, even the new ones, I get this error. I have carried out the update procedure that you recommend (delete the data from the .platformio folder)
How can I solve?

Wiz-IO, 2022-07-05:
yep... will update later
I not have new board for tests and
I need compiler verbose log to look how to construct builder for wifi 

Raspberry Pi support is .... :(
https://github.com/raspberrypi/pico-sdk/issues/897

Wiz-IO, 2022-07-05:
btw: for pico w select board named **WizIO - Raspberry Pi Pico WiFi ( PICOPROBE ... need SDK 1.4.0 )**
board = **raspberry-pi-pico-w**

and select WiFi arch from INI
build_flags =  **-D PICO_CYW43_ARCH_POLL**

Wiz-IO, 2022-07-05:
```ini
[env:raspberry-pi-pico-w]
platform = wizio-pico
board = raspberry-pi-pico-w
framework = baremetal
build_flags = 
    -D PICO_STDIO_USB
    -D PICO_CYW43_ARCH_POLL
```

Wiz-IO, 2022-07-10:
please wait 10-15 mins and reinstall this platform

closed 2022-07-10 by Wiz-IO