Wiz-IO/wizio-pico #108: baremetal serial monitor not working

shablam, 2022-07-17:
Hello, thanks for making this library.
I was using VisialMicro with Arduino and it offers Serial, Serial1 so I can enable Serial1 and get a serial monitor over that. However, I have tried everything I can find and I am unable to get any serial data to transfer in Wizio-pico under VSC. 

I can get Com3 which is Interface 0 to connect on the Serial Monitor plugin under Visual Studio Code but I can't get anything to print there or any other screen. without serial monitor there is not much testing I can do. I have tried many things over some hours and no luck. Can you help?

I have Windows 10.

`
[env:raspberry-pi-pico]
platform = wizio-pico
board = raspberry-pi-pico
framework = baremetal

monitor_speed = 115200
debug_tool = picoprobe
`
that is the basic I have for Platformio.ini.

is there a monitor upload port or speed or something else needed. According to your youtube videos and documentation and other posts this should work. You do not know show your platformio.ini file so I cannot compare it. 

I have tried some other options but nothing helps. I have tried WinUSB libusb-win, libusbK and same for all. I tried PUtty on Com3 but that not work and seems to lock up VSC.

if I use the built in Serial monitor I get: 
"Error: could not open port 'COM3': PermissionError(13, 'Access is denied.', None, 5)"

Nothing else is using the port so I assume something in library changed? Odd that in Arduino I have Serial and Serial1 which allows me to target which one to use. Apparently this is not possible in the Pico-Sdk?  

Thanks for anyone's thoughts. Maybe something needs to be enabled I am not aware of. 

[image]

[image]

[image]

[image]

Wiz-IO, 2022-07-17:
Hi
Baremetal ( here is just a name for sdk ) work as RPI Pico SDK - stdio / **printf()**
need defines:
https://github.com/Wiz-IO/wizio-pico/wiki/BAREMETAL#stdio-printf

Wiz-IO, 2022-07-17:
```ini
build_flags = 
        -D LIB_PICO_STDIO_USB ; and/or
        -D LIB_PICO_STDIO_UART ; and/or
        -D LIB_PICO_STDIO_SEMIHOSTING
```

shablam, 2022-07-17:
Hi, Ok i will try those. I saw that link but I did not know what to put as Foo and Boo are obviously nor real nor is rover and mars I think. Thus I was not making any progress.

shablam, 2022-07-17:
I think I made progress since I last used those flags. one thing seems I maybe had a loose wire as well. Now the flags are actually working. thanks my friend. UART flag did it. I want to integrate TinyUSB so things will change but this is so great! Appreciate your quick reply.

shablam, 2022-07-17:
Can you point me as how to include tinyusb? Have hunted and not sure where to add that in the platformio.ini i assume. plenty of arduino examples but not much with how to add tinyusb into platformio on pico.

shablam, 2022-07-17:
ok it finds tusb.h but not bsp/board.h as no board.h exists anywhere. It exists in the SDK-PICO if downloaded by itself, but not the tinyusb library from what i see.

C:\Users\...\.platformio\packages\framework-wizio-pico\SDK\lib\tinyusb

Wiz-IO, 2022-07-18:
build_flags =  **-D PICO_USB** ; link tyniUSB library

build_flags = -D PICO_USB_HOST load tinyusb as host
or as device
build_flags = -D PICO_USB_DEVICE and/or -D PICO_STDIO_USB ( for printf )
https://github.com/Wiz-IO/wizio-pico/issues/101

if you want to change board ( not tested for all boards )
Edit project include/pico/config_autogen.h

shablam, 2022-07-18:
I know you have not done any host mode USB but I am aimed to figure it out. I see the other post as well. in the code there is "LIB_TINYUSB_HOST". is this a build flag or a define that needs to be set? I can't find anywhere it gets set in the SDK. I am working backwards to try to figure out why some includes are not getting loaded.

Wiz-IO, 2022-07-18:
if substring USB is in definition then tinyusb will be compiled
https://github.com/Wiz-IO/wizio-pico/blob/ee2ddfc49e6bf2ac1204c2e997519bbb03937ad8/builder/frameworks/pico.py#L257

just add in ini 
build_flags = -D PICO_USB_HOST

shablam, 2022-07-18:
Thanks, I did do that. but with the examples I have seems still some issues. Not sure ifthe exmples are bad but am trying to use what is out there.

`
src\hid_app.c:42:49: error: 'HID_KEYCODE_TO_ASCII' undeclared here (not in a function)
 static uint8_t const keycode2ascii[128][2] =  { HID_KEYCODE_TO_ASCII };
                                                 ^~~~~~~~~~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.
*** [.pio\build\raspberry-pi-pico\src\hid_app.o] Error 1
In file included from C:\Users\User1\.platformio\packages\framework-wizio-pico\SDK\lib\tinyusb\src/host/usbh.h:35:0,
                 from src\main.c:34:
C:\Users\User1\.platformio\packages\framework-wizio-pico\SDK\lib\tinyusb\src/host/hcd.h:162:52: error: unknown type name 'hcd_devtree_info_t'; did you mean 'hcd_event_t'?
 extern void hcd_devtree_get_info(uint8_t dev_addr, hcd_devtree_info_t* devtree_info);
                                                    ^~~~~~~~~~~~~~~~~~
                                                    hcd_event_t
compilation terminated due to -Wfatal-errors.
*** [.pio\build\raspberry-pi-pico\src\main.o] Error 1

`

you can see the errors I get there. I assume I need those things.

Wiz-IO, 2022-07-18:
so 
'HID_KEYCODE_TO_ASCII' undeclared here (not in a function)
is example problem ... I not have idea for now and I am too far from home PC to test codes

btw and look in project/inlude tinyusb config file

shablam, 2022-07-18:
at the moment i have no errors with this that you have in your zip file you gave that other person.

`
#ifndef _PICO_STDIO_USB_TUSB_CONFIG_H
#define _PICO_STDIO_USB_TUSB_CONFIG_H

#include "pico/stdio_usb.h"

#if 0 /* usb device for printf() */

#define CFG_TUSB_RHPORT0_MODE   (OPT_MODE_DEVICE)
#define CFG_TUD_CDC             (1)
#define CFG_TUD_CDC_RX_BUFSIZE  (256)
#define CFG_TUD_CDC_TX_BUFSIZE  (256)
#define CFG_TUD_VENDOR          (0)

#else /* usb host */

#define RP2040_USB_HOST_MODE
#define CFG_TUSB_RHPORT0_MODE       OPT_MODE_HOST
#define CFG_TUH_HUB                 1
#define CFG_TUH_HID                 4
#define CFG_TUSB_HOST_HID_GENERIC   0
#define CFG_TUH_HID_KEYBOARD        0
#define CFG_TUH_HID_MOUSE           1
#define CFG_TUH_CDC                 0
#define CFG_TUH_MSC                 0
#define CFG_TUH_VENDOR              0
#define CFG_TUSB_HOST_DEVICE_MAX    (CFG_TUH_HUB ? 5 : 1) // normal hub has 4 ports

#endif

#endif
`

also added 
#define LIB_TINYUSB_HOST 1
in main.c

I will do some more work and report back. thanks so far for the help.

Wiz-IO, 2022-07-18:
then this example worked
https://github.com/Wiz-IO/wizio-pico-examples/tree/main/baremetal/pico-usb-webserver

or find some tinyusb example or project in github

btw: if printf() via USB work ... is HOST and tinyusb is OK

shablam, 2022-07-18:
appreciate the help. I will see if I can get a mouse to connect is first host goal. so glad you made this.

shablam, 2022-07-18:
Yes for the first time ever the mouse LED red light on the bottom lights up! that's a win so far!

shablam, 2022-07-19:
IS there a SerialUSB library for baremetal? I see it for Arduino.

Wiz-IO, 2022-07-19:
is RPI provide library - copy / paste
btw: try printf() / getchar() with
build_flags =     -D PICO_STDIO_USB

shablam, 2022-07-19:
Hi thanks, you mean printf() for serial over USB? I have TinyUSB right now and that over rides RPI I think. looking at options.

Wiz-IO, 2022-07-19:
yep - usb stdio

shablam, 2022-07-19:
I am using picoprobe so that outputs serial of the probe not the target. How would I get that to printf out the target? Currently i can't get the target USB host output to acknowledge the other board with the CH340/1 chip USB to serial i think it is.

Wiz-IO, 2022-07-19:
build_flags = -D PICO_STDIO_USB
retarget stdio ( printf / gerchar ) to USB as pico sdk
https://github.com/Wiz-IO/wizio-pico/wiki/BAREMETAL#stdio-printf

shablam, 2022-07-19:
is there a reason you not using lib_pico_stdio_usb ?

[image]

Wiz-IO, 2022-07-19:
this defines is and/or
look pico sdk documentation

LIB_PIKO_xxxx is for old SDK

shablam, 2022-07-19:
Thanks, I am learning a lot. Never got this deep into C.

shablam, 2022-07-19:
Can I have host mode and -D PICO_STDIO_USB? 

it seems I have to enable Device mode now? 

`
C:\Users\User1\.platformio\packages\framework-wizio-pico\SDK\pico\pico_stdio_usb\stdio_usb.c:172:2: warning: #warning stdio USB was configured along with user use of TinyUSB device mode, but CDC is not enabled [-Wcpp]
 #warning stdio USB was configured along with user use of TinyUSB device mode, but CDC is not enabled
  ^~~~~~~
C:\Users\User1\.platformio\packages\framework-wizio-pico\SDK\pico\pico_stdio_usb\stdio_usb.c:190:42: error: unknown type name 'cdc_line_coding_t'
 void tud_cdc_line_coding_cb(uint8_t itf, cdc_line_coding_t const* p_line_coding) {
                                          ^~~~~~~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.
C:\Users\User1\.platformio\packages\framework-wizio-pico\SDK\pico\pico_stdio_usb\reset_interface.c:103:51: error: unknown type name 'cdc_line_coding_t'
 void tud_cdc_line_coding_cb(__unused uint8_t itf, cdc_line_coding_t const* p_line_coding) {
                                                   ^~~~~~~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.
*** [.pio\build\raspberry-pi-pico\sdk\SDK\pico\pico_stdio_usb\stdio_usb.o] Error 1
Compiling .pio\build\raspberry-pi-pico\sdk\SDK\pico\usb\class\cdc\cdc_rndis_host.o
*** [.pio\build\raspberry-pi-pico\sdk\SDK\pico\pico_stdio_usb\reset_interface.o] Error 1
C:\Users\User1\.platformio\packages\framework-wizio-pico\SDK\pico\pico_stdio_usb\stdio_usb_descriptors.c:40:24: error: 'TUD_CONFIG_DESC_LEN' undeclared here (not in a function); did you mean 'USBD_DESC_LEN'?
 #define USBD_DESC_LEN (TUD_CONFIG_DESC_LEN + TUD_CDC_DESC_LEN)
                        ^
C:\Users\User1\.platformio\packages\framework-wizio-pico\SDK\pico\pico_stdio_usb\stdio_usb_descriptors.c:90:36: note: in expansion of macro 
'USBD_DESC_LEN'
 static const uint8_t usbd_desc_cfg[USBD_DESC_LEN] = {
                                    ^~~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.
*** [.pio\build\raspberry-pi-pico\sdk\SDK\pico\pico_stdio_usb\stdio_usb_descriptors.o] Error 1
`

Wiz-IO, 2022-07-19:
yep
but you need edit default descriptors
https://github.com/Wiz-IO/framework-wizio-pico/blob/main/SDK/pico/pico_stdio_usb/stdio_usb_descriptors.c

shablam, 2022-07-19:
I stil need to be acting as the host. can I have host and device mode enabled? I thought it was not possible.

closed 2022-07-21 by Wiz-IO