Wiz-IO/wizio-pico #114: Serial not working properly on Windows (no scanf or Serial.read)

bastian2001, 2022-10-03:
I am using the arduino-framework and the build flags as follows: 
```ini
build_unflags  = -D BOO
build_flags    =
	-D FOO
	-D LIB_PICO_STDIO_USB
	-D PICO_STDIO_ARDUINO
```
Removing either of the latter two flags results in Serial not working at all.
My project has `Serial.begin(115200)` in it. Normal printf works, even though I did **not** set `retarget` to `true`. Surprisingly, the baud rate is completely irrelevant. But if I actually set `retarget` to `true`, like shown in the docs, printf does not work anymore.

I would like to use the Arduino functions Serial.available and Serial.read or any of their standard C counterparts. Is there a possibility to get something like scanf or Serial.read working? Because as of now, I can only output data to the PC and not send Serial data from the PC to the Pico.

Wiz-IO, 2022-10-03:
PICO_STDIO_ARDUINO is not pico key
LIB_PICO_STDIO_USB enable stdio over USB -> com & is not Serial
SerialN is for hardware urts
for USB - SerialUSB lib

bastian2001, 2022-10-03:
Thank you very much for your clarification, but I still cannot get scanf() or any similar function to work. In fact, when I run scanf, the program stalls completely. Is it possible to get any data from the PC over USB-Serial to the RP2040?

```C
char str[128];
void loop() {
	// put your main code here, to run repeatedly:
	delay (5000);
	printf("running\n");
	scanf("%s", str);
	printf("%s\n", str);
}
```
In the Serial monitor, only "r" is printed, not even the full "running".

Wiz-IO, 2022-10-03:
Arduino not use STDIO ( maybe work, maybe not )
Arduino use Serial.read(), Serial.write() to physical UART or SerialUSB to USB2COM

if you re-target stdio to uart-1 and Arduino use uart-1 as Serial - will be mish-mash

no idea if scanf works...

closed 2022-10-15 by Wiz-IO