Wiz-IO/wizio-pico #115: including Print.h gives namespace error

shablam, 2022-10-21:
Hello, I am back. I have successfully though painfully got TinyUSB working with your wizio-pico. You were right.. i learned a LOT.

I was trying to add in TFT_eSPI which is an arduino library for LCD displays. The kicker is it should work with Platformio but not meant for anything other than arduino platform. 

In TFT_eSPI.h it includes print.h. From there it gets messy:
.platformio\packages\framework-wizio-pico\library\wiring\include/Printable.h:27:1: error: unknown type name 'namespace'; did you mean 'isspace'?

I notice you added in the namespace in the code on print.h and printable.h for avr. the Code i find does not have the namespace. I assume to avoid confusion. 

So the question is do I have to change the compiler to g++ or what is needed so this error will not show up? It is nothing I changed other than include print.h. I dont have to include TFT_eSPI.h, i can just add include print.h to wizio.h and same error. 

Thanks in advance.

Wiz-IO, 2022-10-22:
the idea of "Arduino" is: to work
regardless of the software tools

where do you post print.h ... try Arduino.h

shablam, 2022-10-22:
HI, ok. I ended up moving Ardiuino.h and print.h to main.c. i also renamed main.c to main.cpp. a few other minor changes and at least it compiles. I will be testing soon and will let you know how it goes!  Thanks.

closed 2022-10-25 by Wiz-IO