Wiz-IO/wizio-pico #116: requestFrom in Wire have a bug.

Andres1012, 2022-10-22:
The method requestFrom of Wire Class always return 0, because available() return `_numElems` of RingBuffer.
`int res  = i2c_read_timeout_us(ctx, _slave_address, rx._aucBuffer, size, 0, _timeout_us);`
`if(res < 0)`
    `size = 0;`
res variable get de correct size but the Ring buffer not yet, I recommend this change:
    `rx._numElems =i2c_read_timeout_us(ctx, _slave_address, rx._aucBuffer, size, 0, _timeout_us);`
`if (rx._numElems < 0)`
   ` size = 0;`

Wiz-IO, 2022-10-22:
Thanks !
done
https://github.com/Wiz-IO/framework-wizio-pico/blob/main/arduino/libraries/RP2040/Wire/Wire.cpp#L30

closed 2022-10-22 by Andres1012

shablam, 2022-10-27:
i decided to try arduino instead of baremetal since I could not get tft_espi working. too much headache tying to blend it in. but i get a lot of errors in Platformio.  this one relates to your changes above. apparently there are 2 ring buffer files?

C:\**.platformio\packages\framework-wizio-pico\arduino\libraries\RP2040\Wire\Wire.cpp:30:8: error: 'RingBuffer {aka class RingBufferN<256>}' has no member named '_numElems'

shablam, 2022-10-27:
i added this at the bottom.

`public:
  RingBufferN(void);
  void store_char(uint8_t c);
  void clear();
  int read_char();
  int available();
  int availableForStore();
  int peek();
  bool isFull();
  int _numElems;`

in ringbuffer.h under arduino/arduino folders. error gone. not sure if i fixed it but take a look.

Wiz-IO, 2022-10-27:
Will test soon - waithing boards...

shablam, 2022-10-27:
ya adding that without declaration caused issue. that should fix it.

shablam, 2022-10-27:
you are using really old adafruit GFX files. new ones wont work also. requires additional libraries and creates erors in PIO

shablam, 2022-10-27:
ok, found some other issues. change the MSBFIRST and MSBLAST in SPI.h to SPI_MSB_FIRST and SPI_LSB_FIRST. you have constants.h commented out and that has MSBFIRST LSBFIRST that are not available and causes an error in the compiler if constants.h is enabled. so far that what I see.