Wiz-IO/wizio-pico #119: Picoprobe not working properly on windows

dacoolinus, 2022-11-11:
I am looking to use a picoprobe as a programmer and debugger with wizio-pico. Given the documentation, it would appear all I need to do is to add `upload_protocol = picoprobe` and `debug_tool = picoprobe`. I did this and yet the terminal still attempts to convert the file to a .U2F and attempt to upload it to a Pico mounted as a USB mass. I wondered if the tool-pico-openocd package wasn't being loaded so I added it to lib_deps and still nothing.

here is my platform.ini
```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html
[env:raspberry-pi-pico]
platform = wizio-pico
board = raspberry-pi-pico
framework = baremetal
lib_deps =
  https://github.com/Wiz-IO/tool-pico-openocd.git
  freertos
upload_protocol = picoprobe
debug_tool = picoprobe
build_flags = 
  -D PICO_CYW43_ARCH_POLL

;monitor_port = SELECT SERIAL PORT
;monitor_speed = 115200

```

I am on Windows and if I attempt to use `lib_deps = tool-pico-openocd` then it says it cannot find a package for my system (windows-amd64).

dacoolinus, 2022-11-11:
Uploading via "picoprobe" works but not as anticipated. Might look into why but after digging around the Q&A found a passable workaround.

closed 2022-11-11 by dacoolinus

CHTUZKI, 2023-02-10:
Hello, has your problem been solved? I also encountered the same problem. I use picoprobe, click debug, nothing happens, the terminal prints "Pico USB drive not found.", but my device manager lists picoprobe device,

```
[env:raspberry-pi-pico]
platform = wizio-pico
board = raspberry-pi-pico
framework = baremetal

;monitor_port = SELECT SERIAL PORT
;monitor_speed = 115200

;lib_deps = 

build_type = debug
upload_protocol = picoprobe
debug_tool = picoprobe

board_build.pio = src\hal_rp2040\encoder.pio
```

dacoolinus, 2023-02-10:
@CHTUZKI I definitely found a fix. I wish I could remember off the top of my head but happy to help try and sort out your issue. I think in the end, I had to add `windows-amd` to the tool-pico-openocd's library.json file. Is the only output you get "Pico USB Drive not Found" or does it also give you a more specific error for why it did not try to use openocd?

Wiz-IO, 2023-02-10:
last time it worked...
and I haven't touched the code (more than a year)
I don't have two boards to test at the moment
https://www.youtube.com/watch?v=SdpsmgRp5Co&ab_channel=GeorgiAngelov

TA1DB, 2023-02-10:
> ... click debug, nothing happens, the terminal prints "Pico USB drive not found.",  --- 
are you sure that you're starting a proper debug task, or, are you starting an upload task actually by mistake !

here I am doing debug without any problem but if I start an upload I get the same "Pico USB drive not found." message; this is already documented here :  https://github.com/Wiz-IO/wizio-pico/issues/94#issue-1212325356