Wiz-IO/wizio-pico #120: UnicodeDecodeError:

tcomlab, 2023-01-05:
Hi. Framework generates APPLICATION.uf2 after build normally, but after press upload button get error.

`
Configuring upload protocol...
AVAILABLE: cmsis-dap, picoprobe, uf2
CURRENT: upload_protocol = picoprobe
Uploading...
  Converting to UF2 ( 0x10000000 )
  Wrote 69632 bytes to C:/Users/Raketa/Documents/PlatformIO/Projects/BmRP2040/.pio/build/raspberry-pi-pico\APPLICATION.uf2
*** [upload] UnicodeDecodeError : 'utf-8' codec can't decode byte 0x8d in position 198: invalid start byte
Traceback (most recent call last):
  File "C:\Users\Raketa\.platformio\packages\tool-scons\scons-local-4.4.0\SCons\Action.py", line 1318, in execute
    result = self.execfunction(target=target, source=rsources, env=env)
  File "C:\Users\Raketa\.platformio\platforms\wizio-pico\builder\frameworks\uf2conv.py", line 342, in dev_uploader
    drives = get_drives()
  File "C:\Users\Raketa\.platformio\platforms\wizio-pico\builder\frameworks\uf2conv.py", line 206, in get_drives
    for line in to_str(r).split('\n'):
  File "C:\Users\Raketa\.platformio\platforms\wizio-pico\builder\frameworks\uf2conv.py", line 200, in to_str
    return b.decode("utf-8")
UnicodeDecodeError: 'utf-8' codec can't decode byte 0x8d in position 198: invalid start byte
=================================================================== [FAILED] Took 8.09 seconds ===================================================================
`
[image]

Wiz-IO, 2023-01-05:
Hi
uf2conv.py is not my source ... Microsoft

maybe there is a cyrillic symbol in path
https://github.com/Wiz-IO/wizio-pico/blob/main/builder/frameworks/uf2conv.py#L200

tcomlab, 2023-02-04:
I'm resolved problem

change in uf2conv.py

def to_str(b):
    return b.decode("utf-8")

to

def to_str(b):
    return b.decode("utf-8",errors='ignore')