Wiz-IO/wizio-pico #123: Calling printf("%d", i); without a trailing new line character won't result in an output or a very delayed output

theholypumpkin, 2023-02-01:
Greetings,
I noticed that I can't get printf to print anything except for characters unless I also print a newline character.
With new line character and without  integer overflow

```cpp
int main(void){
    stdio_usb_init();
    while(!stdio_usb_connected());
    for(uint8_t i = 0; i < 255; i++){
        printf("%d\n", i); //this will work just fine
        sleep_ms(100);
    }
    return 0;
```

No new line char and without integer overflow

```cpp
int main(void){
    stdio_usb_init();
    while(!stdio_usb_connected());
    for(uint8_t i = 0; i < 255; i++){
        printf("%d", i); //no output whatsoever.
        sleep_ms(100);
    }
    return 0;
```

No new line and with integer overflow (the loop runs until < 256 so it will wrap around.)

```cpp
int main(void){
    stdio_usb_init();
    while(!stdio_usb_connected());
    for(uint8_t i = 0; i < 256; i++){
        printf("%d", i); //outputs a burst i.e all at once after a long delay
        sleep_ms(100);
    }
    return 0;
/* output (first burst):
0 1 2 3 4 5 6 7 8 9 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 26 27 28 29 30 31 32 33 34 35 36 37 38 39 40 41 42 43 44 45 46 47 48 49 50 51 52 53 54 55 56 57 58 59 60 61 62 63 64 65 66 67 68 69 70 71 72 73 74 75 76 77 78 79 80 81 82 83 84 85 86 87 88 89 90 91 92 93 94 95 96 97 98 99 100 101 102 103 104 105 106 107 108 109 110 111 112 113 114 115 116 117 118 119 120 121 122 123 124 125 126 127 128 129 130 131 132 133 134 135 136 137 138 139 140 141 142 143 144 145 146 147 148 149 150 151 152 153 154 155 156 157 158 159 160 161 162 163 164 165 166 167 168 169 170 171 172 173 174 175 176 177 178 179 180 181 182 183 184 185 186 187 188 189 190 191 192 193 194 195 196 197 198 199 200 201 202 203 204 205 206 207 208 209 210 211 212 213 214 215 216 217 218 219 220 221 222 223 224 225 226 227 228 229 230 231 232 233 234 235 236 237 238 239 240 241 242 243 244 245 246 247 248 249 250 251 252 253 254 255 0 1 2 3 4 5 6 7 8 9 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 26 27 28 29 30 31 32 33 34 35 36 37 38 39 
*/
/* Second burst
40 41 42 43 44 45 46 47 48 49 50 51 52 53 54 55 56 57 58 59 60 61 62 63 64 65 66 67 68 69 70 71 72 73 74 75 76 77 78 79 80 81 82 83 84 85 86 87 88 89 90 91 92 93 94 95 96 97 98 99 100 101 102 103 104 105 106 107 108 109 110 111 112 113 114 115 116 117 118 119 120 121 122 123 124 125 126 127 128 129 130 131 132 133 134 135 136 137 138 139 140 141 142 143 144 145 146 147 148 149 150 151 152 153 154 155 156 157 158 159 160 161 162 163 164 165 166 167 168 169 170 171 172 173 174 175 176 177 178 179 180 181 182 183 184 185 186 187 188 189 190 191 192 193 194 195 196 197 198 199 200 201 202 203 204 205 206 207 208 209 210 211 212 213 214 215 216 217 218 219 220 221 222 223 224 225 226 227 228 229 230 231 232 233 234 235 236 237 238 239 240 241 242 243 244 245 246 247 248 249 250 251 252 253 254 255 0 1 2 3 4 5 6 7 8 9 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 26 27 28 29 30 31 32 33 34 35 36 37 38 39 40 41 42 43 44 45 46 47 48 49 50 51 52 53 54 55 56 57 58 59 60 61 62 63 64 65 66 67 68 69 70 71 72 73 74 75 76  
*/
```
After pasting the output into MS Word and checking the character count (including spaces) I noticed the burst is1024 characters long. So 2^10.

Here also the platform.ini
```
[env:pico-dap]
platform = wizio-pico
board = pico-dap
framework = baremetal

monitor_port = COM8
monitor_speed = 115200

board_build.nano = disable
build_flags = 
    -D LIB_PICO_STDIO_SEMIHOSTING
    -D LIB_PICO_STDIO_UART
    -D LIB_PICO_STDIO_USB
    -D PICO_DIVIDER_HARDWARE
```

I highly assume It's just a wrong build flag but after trying the last two and a half hours to find the right combination of flags without success.

In the coming days, when I find the time I'll try the same code on a personal machine instead of one issued by my employer, checking if the issue is also present there.

Thanks in Advance

Wiz-IO, 2023-02-01:
Hi
Try, project INI
```build_unflags  = -D PICO_STDIO_ENABLE_CRLF_SUPPORT=0```

by default PICO_STDIO_ENABLE_CRLF_SUPPORT = 1

theholypumpkin, 2023-02-01:
Sadly not, exactly the same behavior as before.

Wiz-IO, 2023-02-01:
setvbuf(stdout, NULL, _IONBF, 0);
        setvbuf(stderr, NULL, _IONBF, 0);
        setvbuf(stdin, NULL, _IONBF, 0);

theholypumpkin, 2023-02-01:
Thanks this did work.
I ask for my own sainity and the sake of documentation.
Acording to my understanding those 3 lines set the stdin, stdout and stderr to unbuffered

Else the issue could be closed.
Thanks for the quick help again 👍

Wiz-IO, 2023-02-01:
yep
pico-printf is not wraped and app use gcc-printf

closed 2023-02-01 by theholypumpkin