Wiz-IO/wizio-pico #125: SD card VFS+FAT support

vvveasel, 2023-03-10:
I appreciate that SD card FAT FS support is marked as 'untested', however given that wiki and example exists, it is fair to raise an issue I guess.

I am trying to compile and run SD card [example ](https://github.com/Wiz-IO/wizio-pico-examples/tree/main/baremetal/pico-sd-card) and while `vfs_mount(FATFS_LETTER, &fatfs_ctx);` returns 0 (I guess it is success), the call in the main `fd = open(FILE_NAME, O_WRONLY | O_CREAT | O_TRUNC);` ends up in `File FD = -1 Error: : Invalid argument` debug print.

Any pointers would be greatly appreciated. In the long run I intend to use WIZ-IO with FreeRTOS  -  maybe there is a better approach to enable FAT FS over SD in such configuration?

Wiz-IO, 2023-03-10:
Hi,
I don't have any boards to test at the moment.
This worked
https://github.com/Wiz-IO/wizio-pico-examples/tree/main/baremetal/pico-sd-card

look and SPI driver
https://github.com/Wiz-IO/framework-wizio-pico/blob/main/library/VFS/fatfs_sd.c

vvveasel, 2023-03-10:
Thank you for getting back to me on this.

> I don't have any boards to test at the moment. 

You did a brilliant job here Georgi, echoing others - hate to see you dropping support.
I am happy to send you a set of latest boards if this could help.

> This worked
https://github.com/Wiz-IO/wizio-pico-examples/tree/main/baremetal/pico-sd-card

Yes, this what I am trying.

> look and SPI driver
https://github.com/Wiz-IO/framework-wizio-pico/blob/main/library/VFS/fatfs_sd.c

Well fatfs_sd.c getting compiled... I don't see anything hardcoded or whatever..
Is using a different SPI instance could be a problem? My setup below:
```c
#define FATFS_SPI               spi0
#define FATFS_SPI_BRG           10000000
#define FATFS_SPI_SCK           2 /* SPI1_SCK  */
#define FATFS_SPI_MOSI          3 /* SPI1_TX   */
#define FATFS_SPI_MISO          4 /* SPI1_RX   */
#define FATFS_CS_PIN            5 /* SPI1_CSn  */
```

Wiz-IO, 2023-03-10:
or dont import my FatFS - try compile own
I think there was a SPI driver pico-examples

vvveasel, 2023-03-10:
that makes sense. No worries. I guess it can be closed then

closed 2023-03-10 by vvveasel

Wiz-IO, 2023-03-10:
https://www.youtube.com/watch?v=Lolq46mt-vY

https://github.com/Wiz-IO/wizio-pico-examples/blob/main/baremetal/pico-play.rar

vvveasel, 2023-03-11:
Argh... Apparently that was my silly wiring mistake. Re-tested briefly with your VFS code and it seems to be working. Thanks for pointers.

Wiz-IO, 2023-03-11:
:) nice ...  happy coding